FreeCAD/FreeCAD #14334: New Cam simulator: resizing window does not affect viewing area

ShamanTcler, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I resized the window larger expecting the piece to scale up. The window became larger but the viewing area remained the same.  Only 25% of the window was being utilized.

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.36
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-05-28:
Would you be so kind to provide a screenshot?

maxwxyz, 2024-05-28:
@shaise FYI

shaise, 2024-05-28:
Not yet, but I'll add to the list. I will also check if I can do something that will make window scaling work.

shaise, 2024-07-13:
The option to arbitrarily scale the window is now implemented in my branch.
This feature, however, will be added to main freecad only after 1.0 release since we are in a code freeze.

https://youtu.be/EMYRSip1Nwk

shaise, 2024-07-31:
Update: Since the fully featured simulator solves some other pending issues, I decided to PR it to 1.0 anyway. Hopefully when its merged it will solve this issue. See `Cam simulator feature update` #15597

closed 2024-08-21 by sliptonic