FreeCAD/FreeCAD #14342: UI: Visibility toggle should be possible with active tools (eye icon)

maxwxyz, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently https://github.com/FreeCAD/FreeCAD/pull/12298 implemented an eye icon to toggle visibility of objects by directly clicking on the respective eye icon. The state of visibility / hidden is displayed with the icon as well. This setting is disabled by default.

If currently a tool or task is active and you need to hide an object to reference or select / see features you currently have to cancel the action, select the object and hide it manually. The visibility eye icon lets users also toggle visibility when an active tool or task is open.
E.g. using Sketcher to select external geometry: You can hide the objects which are in the way by directly clicking on the eye icon. Without the eye icon visible, you need to cancel the tool, select the object and hide it and activate the tool again.
Also in FEM dialogs when you want to select features and need to hide a mesh or constraints, it is possible to do this on the go.
[image]

The concept itself is very understandable as it is a known behavior in other CAD applications (NX, Fusion360) but also other related apps like KiCad, Blender, Inkscape,...  This should also enable new users an easy and quick way to toggle visibility while also allowing to toggle it during active tasks and improve UX which is not possible with the current default setting.
The UX could be further improved on by e.g. clicking and dragging down the tree on the eye icons to quickly hide or unhide multiple objects (as known from layer based applications like Inkscape, Gimp, Adobe).

DWG discussion result:
- Eye icon setting is active by default (needed as toggle, nice to have as indicator)
- Eye icon should also be greyed out when object is hidden
- Text is greyed out when object is hidden, regardless of the setting of the eye visibility
- Icon is only visible on hover or in greyed out state (object is hidden) On hover and toggle it changes back to black.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NomAnor, 2024-05-28:
> * Icon is only visible on hover or in greyed out state (object is hidden) On hover and toggle it changes back to black.

Should this be configurable? When the icon is not visible, I can't directly move my cursor over it. When I want to hide a visible object, I first have to move the mouse over the object and then I have to move it over the eye icon.

If the icon preference is kept, we could change it to a combobox with three modes:
1. Always visible
2. Auto-hide
3. Always hidden

maxwxyz, 2024-05-28:
@NomAnor I've updated the OP. With hover I meant hover the combobox / the entire tree item. So by default the icon is invisible and when you hover anywhere on the tree item, it is shown. It is implemented like this in OnShape iirc.

microbearsmb, 2024-05-28:
I agree that all three modes should be available to the user, with the onHover behavior as default.

However, @NomAnor, I don't think there's much of an issue with having to move your mouse multiple times for the onHover behavior. You would move your mouse from wherever it is onscreen until it is hovering over the icon of the object you want to hide, and the icon/text would jump to the right, revealing the eye right beneath your mouse.

NomAnor, 2024-05-28:
> With hover I meant hover the combobox / the entire tree item.

You mean the entire tree widget/panel? That would be better. So the icons for all tree items show up as soon as the mouse cursor enters the tree area.

> I don't think there's much of an issue with having to move your mouse multiple times for the onHover behavior

This might mess with muscle memory because the "layout" changes. It's just something to keep in mind. I will test that if its implemented.

@maxwxyz You can assign me to the issue. Most points look relatively easy to implement.

obelisk79, 2024-05-28:
what combobox are you guys talking about?

obelisk79, 2024-05-28:
> You mean the entire tree widget/panel? That would be better. So the icons for all tree items show up as soon as the mouse cursor enters the tree area.

This would result in too much visual change on screen at once and would become a distraction. It definitely should only display on a per-treeObject basis.

Reqrefusion, 2024-05-28:
If there's anything uglier than an eye, it's a comboBox.

Reqrefusion, 2024-05-28:
It could be something more abstract, like the one in the picture.
[image]

NomAnor, 2024-05-28:
> what combobox are you guys talking about?

Good question, I'm now also confused :D A combobox for me is a special kind of UI widget (QComboBox), I thought @maxwxyz meant the combo panel.

> It could be something more abstract, like the one in the picture.

I like the eye. I think it currently fits with the style of the other icons. 
in #12341 are some examples of other applications that use an eye for visibility. That seems to be a well know symbol.
There was an alternative icon in https://github.com/FreeCAD/FreeCAD/pull/12298#issuecomment-1958500666, but it didn't work with the overlay style.

maxwxyz, 2024-05-28:
combobox was mentioned on Discord. I meant the entire tree item width.
I also like the current icon.

microbearsmb, 2024-05-28:
I agree with the current icon; as per the discussion on Discord, it fits nicely with the rest of the UI at the moment. One comment though - should there be a little more space between the eye and the icon of the object in the tree view? It seems a little too close as it stands.

Reqrefusion, 2024-05-28:
> > what combobox are you guys talking about?
> 
> Good question, I'm now also confused :D A combobox for me is a special kind of UI widget (QComboBox), I thought @maxwxyz meant the combo panel.
> 
> > It could be something more abstract, like the one in the picture.
> 
> I like the eye. I think it currently fits with the style of the other icons. in #12341 are some examples of other applications that use an eye for visibility. That seems to be a well know symbol. There was an alternative icon in [#12298 (comment)](https://github.com/FreeCAD/FreeCAD/pull/12298#issuecomment-1958500666), but it didn't work with the overlay style.

> > what combobox are you guys talking about?
> 
> Good question, I'm now also confused :D A combobox for me is a special kind of UI widget (QComboBox), I thought @maxwxyz meant the combo panel.
> 
> > It could be something more abstract, like the one in the picture.
> 
> I like the eye. I think it currently fits with the style of the other icons. in #12341 are some examples of other applications that use an eye for visibility. That seems to be a well know symbol. There was an alternative icon in [#12298 (comment)](https://github.com/FreeCAD/FreeCAD/pull/12298#issuecomment-1958500666), but it didn't work with the overlay style.

I'm glad you liked it. I believe it will make it look less busy than the current icon. Especially if we anticipate that this icon will appear a lot.

NomAnor, 2024-05-28:
> I'm glad you liked it.

I'm sorry, I worded that weird. I meant the current icon. Your icon is a bit too abstract for my taste,  but it might fit with a more modern style. 

It's true that the icon is repeated often on screen and too much color can distract from the more important icons. The black and white example from @marcuspollio was not bad. Maybe the black parts could be thinner. But its difficult to make an icon useable in all cases especially with the overlay style.

NomAnor, 2024-05-30:
I implemented all points in the linked PR. Please test it.

@obelisk79: I don't like the hover behaviour, but maybe I misunderstood how it should work.

NomAnor, 2024-05-30:
[image]

The jumping of the icons on hover is really annoying. Only way to prevent it is to show an empty space instead of the icon.

https://github.com/FreeCAD/FreeCAD/assets/8003824/8b5f7db7-315e-4436-b0a2-c266bf4225e8

Reqrefusion, 2024-05-30:
> ![preference](https://private-user-images.githubusercontent.com/8003824/335377097-d6622e46-5c53-48c1-8453-0930b753373a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcxMDY4MDIsIm5iZiI6MTcxNzEwNjUwMiwicGF0aCI6Ii84MDAzODI0LzMzNTM3NzA5Ny1kNjYyMmU0Ni01YzUzLTQ4YzEtODQ1My0wOTMwYjc1MzM3M2EucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MzBUMjIwMTQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzRhNWRmNTkzNDc5MThlYTQ1NTgyZGE4OWY3MDJiODg4MzZjMjg2NTE1YjFlNWJiN2QwZmE0MWFhNjg5YWZkNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pQ6WHsWXMKWIsjQqwvAoXyaIgpapmFbhOtRPr8OnsJU)
> 
> The jumping of the icons on hover is really annoying. Only way to prevent it is to show an empty space instead of the icon.
> hover.mp4

When you hover over this shape, it slides to the right, which is annoying and not recommended in UX. Because the this behavior deceives the user and misses the thing he wants to click on when he clicks on it. I recommend finding a solution to this.

obelisk79, 2024-05-31:
For non hover state, try using a file transparent icon, that should make spacing easier

NomAnor, 2024-05-31:
This is how it looks with using a blank space when not shown:

https://github.com/FreeCAD/FreeCAD/assets/8003824/b8ce1704-7f09-465e-921b-e25bc5aaa88c

microbearsmb, 2024-05-31:
Better... would it be an improvement to have the arrows auto-hide and show when hovered too?

maxwxyz, 2024-05-31:
> Better... would it be an improvement to have the arrows auto-hide and show when hovered too?

Don't think so. 
It already looks off with that space next to the arrows if no eye is visible, which breaks the hierarchy IMO.
Should this really be the default or just always visible?

NomAnor, 2024-05-31:
> It already looks off with that space next to the arrows if no eye is visible, which breaks the hierarchy IMO.

I don't like the empty space, too. 

There are also ideas to have the transparency toggle available through the eye icon. This might expand into more icons in the future. With that in mind, I would much prefer the approach of blender and have additional icons in separate columns. 
[image]
But this is a discussion for a separate issue.

I personally prefer the "Always visible" mode.

As far as this issue is concerned, is there anything missing? 
I will change "Show visibility icon" to "Visibility icon" in the preference. 
Is the wording of the choices ok?
What mode should be default?

PaddleStroke, 2024-05-31:
The auto-hide would be extremely annoying imo. That means that all the time you have the objects jumping back and forth when hovered. And not only when you want to toggle visibility, but always. If you try to select an object quickly, you are likely to select the eye instead.

I think what would be acceptable, is to have the eye icons drawn on the right side of the label with auto-hide. Then it wouldn't clutter the tree too much. And when you hover an item you have the eye that appears.

maxwxyz, 2024-05-31:
@PaddleStroke yes it's probably better looking but you cannot toggle visibility currently without the icon visible with active tasks. In a sketch with an active tool to get e.g. the right external geometry selected you sometimes need to hide objects zhat are in the way. When any tool is active, space bar does not work. Also on other tasks or in FEM when you try to select features that are hidden by the mesh or other objects the toggle in the tree is the only thing working to disable/enable the visibility without cancelling your current action.

obelisk79, 2024-05-31:
> Better... would it be an improvement to have the arrows auto-hide and show when hovered too?

I would say not. Those lines and arrows indicate hierarchical relations between tree objects and should be always visible.

obelisk79, 2024-05-31:
Perhaps just having the eye icon be on the right of the tree text would be more concise and consistent, even though it may sacrifice *some* consistency in eye icon alignment. Then the hover behavior would likely make more visual sense.

maxwxyz, 2024-05-31:
I'm not sure about the eye at the end. You would constantly need to horizontally scroll in the tree view. Especially the nested tree with Part WB features or when working with assemblies and also longer part names this would be a PITA to hover the tree view, horizontally scroll so far until you see the eye icon for the object you want to hide and then click it. Then scroll back so you can see the other names.

NomAnor, 2024-05-31:
> Perhaps just having the eye icon be on the right of the tree text would be more concise and consistent, even though it may sacrifice _some_ consistency in eye icon alignment.

I would use separate columns as in the blender example. Having separate columns for the icons would align all icons and be more extendable in the future. I don't know how easy it is to add a new column and if that would work with the overlay style, I will have a look but no promise. That is something that will most likely not be possible to implement before Monday for the feature freeze (because of the scrolling problems maxwxyz mentioned).

The current implementation just appends the eye icon on front of the real icon and uses that as the icon in the tree, they are not two differenct icons.

I'm not sure what changes are allowed after June 3. Moving the icon into it's own column might be okay because the feature itself will not change.

The only changes that should probably be commited before the feature freeze are the default for the preference and what modes are available, everything else can probably wait until June 24 when the UI freeze begins.

maxwxyz, 2024-05-31:
would it be possible to have the eyes at the most left position (black dots) in the image?
[image]

Reqrefusion, 2024-05-31:
Having the eyes in different places like this definitely creates a problem. They should be in a single column at the bottom. Whether the column is on the right or left is a matter of preference, but I would put it on the left.

obelisk79, 2024-05-31:
My suggestion is a left column as the tree view is left aligned, and this would minimize distance between the icon and the content of each tree item, this is particularly critical in overlay mode.

kadet1090, 2024-05-31:
This would be quite hard to implement to work with both Qt5 and Qt6 as the standard TreeWidget gives minimal (or simply no) control over this area.

NomAnor, 2024-06-01:
> This would be quite hard to implement to work with both Qt5 and Qt6 as the standard TreeWidget gives minimal (or simply no) control over this area.

I looked into it and it seems to be possible. You can control the column which shows the hierarchy arrows with `QTreeView::setTreePosition()`:
[image]

We either have to shift all columns or we might be able to use `QHeaderView::moveSection()` to only visually move them and keep the current indexes. Then the first column can show the eye icon with a fixed width.

I will try to implement this, if that's the direction we want to go. But I'm not comfortable implementing it by Monday.

I suggest keeping the color (gray) changes and having it on by default. I will then submit the PR.

The hover behaviour and the position need more discussion. If it's ok for the maintainers, we can try to submit these changes by June 24 for the UI freeze.

NomAnor, 2024-06-02:
PR is submitted for review without the hover effect. I will investigate the additional icon columns in the following days.

NomAnor, 2024-06-02:
This is a test, how it would look with the icon in an extra column before the items:
[image]

This needs proper integration into the tree code. But these are a couple of possible features this would allow:
* Allow more utility icons, either all in one column or one column per icon
* Allow the user to configure the column order (icons on the left or right for example)
* Allow the user to hide some icons/columns
* Cleaner separation between the feature icon and the utility icons
* Probably easier handling of click events on the icons

pierreporte, 2024-06-02:
@NomAnor With your last test, we could already have the checkbox there. It’s closer and closer to the Blender tree view shown in https://github.com/FreeCAD/FreeCAD/issues/14342#issuecomment-2141676571.

Reqrefusion, 2024-06-02:
> This is a test, how it would look with the icon in an extra column before the items: ![visibility](https://private-user-images.githubusercontent.com/8003824/335884261-edebaebd-2323-47bd-abfc-31713c5073e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTczNTg4NzQsIm5iZiI6MTcxNzM1ODU3NCwicGF0aCI6Ii84MDAzODI0LzMzNTg4NDI2MS1lZGViYWViZC0yMzIzLTQ3YmQtYWJmYy0zMTcxM2M1MDczZTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDJUMjAwMjU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGE3YjQ0Mzc3ZmM3YmE2ZWI4ZDRmNjhjNTQyNDU5MTg3ZGI5YzZiM2Y1NmVmODYwYzJhZDVjNTMwNWJiODBkMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.B2kZJoDYpTwWl9H8eB_C9QU1u1lBrL3fTPLWCDEOngk)
> 
> This needs proper integration into the tree code. But these are a couple of possible features this would allow:
> 
>     * Allow more utility icons, either all in one column or one column per icon
> 
>     * Allow the user to configure the column order (icons on the left or right for example)
> 
>     * Allow the user to hide some icons/columns
> 
>     * Cleaner separation between the feature icon and the utility icons
> 
>     * Probably easier handling of click events on the icons

I thought it would look like this when it was on the left. It is quite difficult to understand which object the relevant eye belongs to. This problem can be solved in different ways. Having the eyes on the right is one of them. But I think it's still an improvement. Congratulations, great job.

FlachyJoe, 2024-06-02:
> It is quite difficult to understand which object the relevant eye belongs to.

It's better if we reduce the level spacing. Half the current size seems ok:
[image]

pierreporte, 2024-06-02:
@FlachyJoe Reducing the identation makes it even harder than it is currently to correctly identify the level of each object. A better solution would be to highlight the whole line at hover.

FlachyJoe, 2024-06-02:
@pierreporte Code editors solve this with thin lines
[image]

We could do the same and arrow could then move
[image]

pierreporte, 2024-06-02:
Lines are not present in the new default theme (though I asked @MisterMakerNL to add them). There is still going to be a problem for deep trees, while a hover effect solves the problem for every case.

obelisk79, 2024-06-02:
None of these solutions in the recent screenshots look particularly good and will be even more problematic in overlay mode. Just leave the current merged behavior and change the default preference. I'll happily disable the eyes locally. There isn't likely to be a good solution, just "less crappy" compromise.

kadet1090, 2024-06-02:
Hover while definitively needed is also non-ideal as you still won't be able to see at a glance which element is hidden, locked or anything like this without hovering. Alternating rows would help a lot there but will look somewhat strange especially for the overlay.  I second Obelisk here - I doubt that we can provide a good solution with current tree design and limiting factors. Maybe there is a better solution but it will probably require a larger remodel of the tree view.

Reqrefusion, 2024-06-02:
It would be much better if this feature was turned off by default until it becomes more understandable. I think finding the right look is not possible with UI Freeze so close. It's still a must have feature.

MisterMakerNL, 2024-06-03:
> @pierreporte Code editors solve this with thin lines [...]

Makes no sense since the lines are thereto point to the end of the command. Our tree doesn't have ends. The example also goes from part design function too spreadsheet. I guess this function is more focused towards BIM users. And normally used in combination with layers.

FlachyJoe, 2024-06-03:
> Our tree doesn't have ends.

Each level ends at level up.

MisterMakerNL, 2024-06-03:
I think this is hat users want:
[image]

MisterMakerNL, 2024-06-10:
We btw also have an eye icon with a cross in the addon manager:
[image]

kadet1090, 2024-06-10:
This is more coherent with the current icon style that freecad uses, but it would be probably distracting

closed 2024-06-17 by chennes