FreeCAD/FreeCAD #14344: Improving the “description” of an object in the view tree.

kaiwas, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. At the moment, the “description” field displayed in the view tree does not correspond to the name of the property. In the tree view it is “description”, and in the properties panel it is “label2”.
2. To display the description in the view tree, a limited zone is used. In addition, for this column to appear, you need to activate this setting. As a result, this opportunity is difficult and inconvenient to use.
(gif image)
[image]

Offer.
1. In the title "description" change to "label2". And leave the current behavior as is. This will allow you to continue to use the second-level heading when the main heading is missing.
2. Add a “description” text field to the object properties. The text from this field could be displayed when hovering over an object in the view tree (if this property is filled).
It would be possible to write fairly long descriptions of an object, which would be useful for making detailed notes regarding the object. Can be compared to comments in programming. Imagine that you are sharing a file with someone. Such information could be very useful in this case. It is better if this field is always available immediately, without additional activation.
[image]

Related tasks.
At the moment, a pop-up message when hovering over an object in the view tree is already available. But it is only displayed if an error occurs. It might be better to change the error color to red and make the “description” field blue or green.

[image]

### Full version info

```shell
Any
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
There are multiple ideas described in this issue.
I think the first step would be to rename label2 to Description.

kaiwas, 2024-05-30:
> I think the first step would be to rename label2 to Description.

Maybe. But I think that label 2 should remain in the view tree (and therefore in the panel).
This is a second level heading that is sometimes necessary. For example, as a marker.
That is, label 2 in the view tree and label 2 in the properties panel.

And the “description” field is another level. And it is not visible in the tree view directly, but the description is displayed when you hover the mouse.
I hope I conveyed the idea clearly.
[image]

maxwxyz, 2024-05-30:
IMO the default should just display the name in the tree. Otherwise the horizontal space is too limited. Especially with larger part names, assemblies and nested objects (like Part WB does) I already have to constantly scroll in the tree.

kaiwas, 2024-05-30:
> IMO the default should just display the name in the tree. Otherwise the horizontal space is too limited. Especially with larger part names, assemblies and nested objects (like Part WB does) I already have to constantly scroll in the tree.

I just want to give an example of using label2.
I displayed the window dimensions in these cells. It is very comfortable.
In my opinion, label2 should remain in the same place (this is not a “description”)
[image]

yorikvanhavre, 2024-05-30:
> I think the first step would be to rename label2 to Description.

+1 to this. This would make most sense and I don't think there is any bad side effect to it. BIM objects have a Description property already (which comes from IFC, all IFC objects have a Description property), but that would be totally compatible with the idea of having it displayed in the tree. AFAIK no other FreeCAD object has a Description property, and I suppose if there were that would be exactly the same purpose: a more lengthy description of the object.

kaiwas, 2024-05-30:
> +1 to this.

Either you didn't understand my idea correctly, or I don't understand you.
I propose to add another entity that is visible only when you hover over the object. This allows you to make a detailed description that will not take up space in the view tree.
In this case, leave label2 (as it is). Just change the name in the view tree to label2 (now it’s “description” in the title). This feature is also sometimes required. For example, there you can specify keywords or some object parameters (for example, the dimensions of a part).

Label two in the object properties is a text string (as before).
The description in the properties should be a text field so that you can conveniently write a large, detailed description."

It seems my idea has been turned upside down (((

yorikvanhavre, 2024-05-30:
But I don't understand why one would need both Label2 AND Description? Would anyone want to give an object 3 different descriptive infos, a Label, a Label2 and a Description? There would then be a short name, a long name, and an even longer name?

In any case Label2 is not very informative... Then it should maybe become LongName (BTW IFC also has a LongName property)

kaiwas, 2024-05-30:
> But I don't understand

Fine. I'll try to show why this might be needed.
Using a window as an example (but it can be any object)
There are three entities.
1) Main object name (no changes)
2) Label2 (the second name of the object. This is a very short description that takes up almost no space in the view tree. For example, you can display the dimensions of a window there.)
3) Pop-up description (a possibly long phrase that cannot be displayed in the tree view area. is shown
pop-up if not empty)
This item can be very useful if, for example, the model is transferred to another person. He will be able to read a long description if he does not understand what this object is needed for (sketch/wall/window/pad/extrusion/whatever).

Each of these entities is useful and necessary.

[image]

matt-taylor-git, 2024-10-03:
Created https://github.com/FreeCAD/FreeCAD/pull/16990 to rename Label2 to Description. 

Now the property field matches the name in the Tree View field. This is my first contribution. Please let me know if I've missed any steps.

kaiwas, 2024-10-04:
> Created #16990 to rename Label2 to Description.

Excellent!
It would be great if the next step would be a new field with a "floating big description" of the property::string List .