FreeCAD/FreeCAD #14347: Problem: New Start page [AppImage]

kaktusus, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The layout of the icons looks incorrect. The text of the icons cannot be read.

https://forum.freecad.org/viewtopic.php?style=4&p=761903#p761903

https://github.com/FreeCAD/FreeCAD/issues/13991#issuecomment-2109434675

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.6.37
  * OpenDark 2024.3.13
  * Render 2024.1.23
  * Alternate_OpenSCAD 1.0.0
  * Plot 2022.4.17
  * kicadStepUp 10.23.2
  * dodo 1.0.1
  * Reinforcement
  * Rocket 3.3.0
  * pcb 6.2023.1
  * freecad.gears 1.2.0
  * woodworking 0.21.2.33771
  * Marz 0.1.5
  * kicadStepUpMod 11.1.5
  * Movie.backup1716816115.789322 2024.2.15 (Disabled)
  * Movie 2024.2.15
  * fasteners.backup1716822852.9058104 0.5.21 (Disabled)
  * fasteners 0.5.21
  * sheetmetal.backup1716822856.2948024 0.4.14 (Disabled)
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

None

### Anything else?

The following releases:

[image]

[image]

[image]

Such an abnormal condition exists until now.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
I cannot reproduce on Windows with the latest weekly, even when resizing the window.
[image]

kaktusus, 2024-05-30:
Are you using OpenDark?

JohnOCFII, 2024-05-30:
> Are you using OpenDark?

Not currently, no.  Just using the Classic (default) theme for now.  (I got myself all tangled up so used the nuclear option of deleting my user.cfg to get back to a known starting point).

maxwxyz, 2024-05-30:
I'm using OpenTheme as OpenDark was depreciated.
Does it only happen to you with OpenDark? Could you uninstall and test with OpenDark in OpenTheme?

kaktusus, 2024-05-30:
This seems to have helped but the problem is not completely solved.
I have the impression that the button text uses light ink. When I switch the theme in OpenLight everything is visible.

[image]

[image]

maxwxyz, 2024-05-30:
Then this seems to be an issue of the theme, not FreeCAD. You can copy the issue to https://github.com/obelisk79/OpenTheme/

MisterMakerNL, 2024-05-30:
> I cannot reproduce on Windows with the latest weekly, even when resizing the window. [...]

Switch between stylesheets a couple of times and it will break.

80avin, 2024-06-06:
Happens with me on Linux when I've installed freecad as snap.
Switching themes is breaking the dimensions of start tiles, colors are also off and no thumbnails for the example projects. 
I tried clearing up everything and reinstalling but no gains.

JohnOCFII, 2024-07-01:
I've seen this on macOS as well.  I see this when switching between various themes (Class, light, OpenLight, etc.). Sometimes I just hit apply (and/or OK) sometimes I restart FreeCAD.  Long and short of it is I think something in the UI-related themes or overlays or something slams on top of some other settings that shouldn't be related...

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.6.22
```

[image]

closed 2024-07-15 by chennes