FreeCAD/FreeCAD #14348: New Cam Simulator displays only blank image

JohnOCFII, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After pressing the PLAY button, a new window is opened, with  a static blue and white image.  No simulation is shown.  This is on macOS M1/arm.
[image]
[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
@shaise FYI

shaise, 2024-05-29:
unfortunately I do not have a mac-os to debug it

JohnOCFII, 2024-06-02:
I'd guess it is something QT related.  Maybe someone else with experience with QTs video routines might have an idea.  I ran this on Debian Linux, so I have a sense of what it should be doing (although some tool tips would be a nice addition).

Here's my sample file. 
[CAM Test.FCStd.zip](https://github.com/user-attachments/files/15523954/CAM.Test.FCStd.zip)

shaise, 2024-06-02:
@JohnOCFII Yes there are some issues I found in Linux with QT. I have built a Ubuntu machine specially to debug it. The is a waiting PR for that, hopefully it solved Linux issues. Regarding MAC I have no access to such machine.
Here is how it looks on windows:
[image]

boboxx, 2024-06-05:
I have the same issue on FreeCAD_weekly-builds-37645-conda-macOS-arm64-py311.dmg

shaise, 2024-06-05:
Unfortunately I dont have access to a MacOS system, so its hard for me to debug.

boboxx, 2024-06-05:
yes I understand, I just wanted to repport that this seems to be a wide issue and not specific to one user.

I don't have python experiance and very limited on how to setup a developement environement to contribute to this problem but I will be happy to test :)

shaise, 2024-06-05:
What type of Mac system you have? (M chip based, intel based etc...)

boboxx, 2024-06-05:
Apple M1 Pro, running Sonama (14.5)

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenTheme 2024.5.16

xtemp09, 2024-06-05:
Can it be scaling issue? @JohnOCFII, could you vary `QT_SCALE_FACTOR`? [^1]

[^1]: https://doc.qt.io/qt-5/highdpi.html

JohnOCFII, 2024-06-09:
> What type of Mac system you have? (M chip based, intel based etc...)

As mentioned above - I've got an M1/arm-based 16" Mac Book Pro.  I use the default resolution of 1728x1117.

I see your change hasn't been merged yet, but I can build locally, so I'll see if I can insert your change and see if it makes a difference.

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git)
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

JohnOCFII, 2024-06-09:
> Can it be scaling issue? @JohnOCFII, could you vary `QT_SCALE_FACTOR`? [1](#user-content-fn-1-865a0e58ab5cc0a421682f81d6658fb8)

I read up on that, but not sure where to check/change that value.

> 
> P.S. I also noticed on the first screenshot the buggy _NaviCube_ and "Path Simulator" is too small. What is your screen resolution? Does other _OpenGL_ stuff in _FreeCAD_ work?

 I've got an M1/arm-based 16" Mac Book Pro. I use the default resolution of 1728x1117.  The older cam simulator works.  Can you point me at some other FreeCAD function that uses OpenGL so I can test?
 
> 
> P.P.S. Does _Use software OpenGL_ option in _Preferences_ change anything? Does _Mesa_ work?

Changing to `Software OpenGL` has no change in behavior.

As far as Mesa -- are you referring to this?  https://www.mesa3d.org 

According to this page, it should run, but I have not tried to build it. https://docs.mesa3d.org/macos.html

shaise, 2024-06-09:
I have placed a PR (#14513) that might solve it. Waiting for it to be merged.

xtemp09, 2024-06-09:
> Can you point me at some other FreeCAD function that uses OpenGL so I can test?

Edit →‬ Preferences, Display →‬ Light Sources

Also, could you tweak the option Display → ‬3D View → Anti-Aliasing? Does anti-aliasing influence view? You can have two documents opened with different anti-aliasing settings to compare.

Of course, there is radial/linear gradient, but it seems to be working, according to the screenshot.

JohnOCFII, 2024-06-09:
> I have placed a PR (#14513) that might solve it. Waiting for it to be merged.

I manually edited `DlgCAMSimulator.cpp` with the changes in your PR, and your simulator now works on MacOS!  So - once your PR is accepted and a new weekly version is out there, it should work for others as well.

[image]

[image]

[image]

shaise, 2024-06-10:
> I manually edited `DlgCAMSimulator.cpp` with the changes in your PR, and your simulator now works on MacOS! So - once your PR is accepted and a new weekly version is out there, it should work for others as well.

Thanks!
This is good to hear as the fix was just a guess...

xtemp09, 2024-06-10:
I'm very happy to hear that the _New CAM Simulator_ (_Path Simulator_, according to the screenshot) works fine now, but staring at the screenshots provided, I can say that something really bad with _OpenGL_ and stylesheets takes place. Just look at the _NaviCube_. What are the wide lines on the left? Why does the panel one the right have a frame?

UI experts definitely ought to correct this.

shaise, 2024-06-10:
@JohnOCFII , can freecad be compiled on MacBook Air M1?

JohnOCFII, 2024-06-10:
> I'm very happy to hear that the _New CAM Simulator_ (_Path Simulator_, according to the screenshot) works fine now, but staring at the screenshots provided, I can say that something really bad with [...]

As to the UI -- yeah, not sure what the controlling factors are.  I also run FreeCAD in a Debian 12 VM on this same MacBook Pro 16" laptop, and in comparison, the UI looks like Easy Eye.  Most elements appear much larger.

JohnOCFII, 2024-06-10:
> can freecad be compiled on MacBook Air M1?

That should work fine.  Maybe slightly slower compilation as the Air likely has less memory, and fewer CPU cores, but overall, still very speedy.

shaise, 2024-06-18:
@JohnOCFII, @boboxx ,
The latest build should have the fix. please test and let me know.

JohnOCFII, 2024-06-18:
> The latest build should have the fix. please test and let me know.

Yes, the new simulator seems to work properly in build 37819

closed 2024-06-30 by JohnOCFII