FreeCAD/FreeCAD #14350: Pressing RETURN/ENTER in a field in the property view causes 3D main window to switch to start window

JohnOCFII, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After editing the Z position by entering a number and pressing RETURN, the main 3D view window for the current model is replaced with the Start Page.  See images below:
[image]

[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
I cannot reproduce on Windows.
It seems you have the start page open in the first window as well. Does it also open if you close the start page manually?

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

joshualibrarian, 2024-05-29:
I am on MacOS and this problem occurs for me constantly and is somewhat maddening!  I'll be editing a property, and I'll press enter, and the view suddenly switches to a different file!

joshualibrarian, 2024-05-29:
I can describe a very clear way to reproduce:

FreeCAD-0.22.0-37436 on MacOS Sanoma 14.5
(but this issue has been occurring for me for a while, with several versions of each)

1. Start FreeCAD and create two new files.  (I have my system set to no Start Window)
2. In Part workbench, create a Cube in the second of the two files.
3. In the Property View, edit the Length (or any of the three) property in the Data tab with a different value
4. Press enter, directly from editing the value.

Immediately after pressing enter, I find that I am in the state shown in the screenshot, where the cube data is still
showing in the Property view, but the active file is the empty one, and the 3D view is empty (showing the empty file).

[image]

JohnOCFII, 2024-05-29:
@maxwxyz  It would not surprise me if this is macOS specific.  If the Start window is closed, after pressing `Return/enter` after editing a field, the main 3D window will switch to whatever is the first open document in the left (from left to right).

[image]

maxwxyz, 2024-05-30:
Is this only happening in the property view or also in other panels like the tree view, task panel, or when constraining a sketch?

joshualibrarian, 2024-05-30:
> Is this only happening in the property view or also in other panels like the tree view, task panel, or when constraining a sketch?

For me, yes, this only happens in the property view.  When I double-click on that cube and edit those values in the Task Panel, I hit enter and it updates the cube without switching files.  Constraining sketches also seems fine, I hit enter and don't leave the sketch or the file.  It's only the property view so far as I can tell.

maxwxyz, 2024-08-14:
Could be the same as https://github.com/FreeCAD/FreeCAD/issues/15491 but this was also an issue on Linux, not only macOS

chennes, 2024-08-18:
The offending line is here:
https://github.com/FreeCAD/FreeCAD/blob/95c514fca3190c7b56c40f2fc4005a7337f64a56/src/Gui/propertyeditor/PropertyEditor.cpp#L264

That's just a normal Qt call to close the editor, but commenting it out eliminates the switch to the first open tab. So either there is some subtlety we are missing in calling it, or there is a very persistent Qt bug (this occurs in all Qt versions I have tested on MacOS, up to and including 6.7). @wwmayer any ideas for how we could work around this?

wwmayer, 2024-08-18:
> That's just a normal Qt call to close the editor, but commenting it out eliminates the switch to the first open tab.

I don't know if this is a Qt bug. In v0.17 and older a persistent editor was directly closed when another item got the focus. At that time there were not known such issues.

Starting with 3a81bd510368 and many follow-up changes closing the persistent editor was delayed. This caused several problems like flickering issues, crashes, ... For all the appearing problems always a workaround was implemented but I wouldn't be surprised if all this is misusing the Qt API so that things don't behave as they are supposed to do.

chennes, 2024-08-20:
@JohnOCFII do you have the ability to compile FreeCAD? If so, can you test PR #15944 to confirm that it's fixing this bug for you? I works for me, but I want to make sure before we include such an ugly hack.

JohnOCFII, 2024-08-20:
> @JohnOCFII do you have the ability to compile FreeCAD? If so, can you test PR #15944 to confirm that it's fixing this bug for you? I works for me, but I want to make sure before we include such an ugly hack.

I do -- but I'm out of town until tomorrow evening (Wednesday - US Central time), but can try to build it and test at that time.

maxwxyz, 2024-08-20:
it fixes the bug for me

JohnOCFII, 2024-08-22:
Well...  The operation in this issue **does** now work correctly with this change... BUT it ALSO seems to work in the weekly build (from about a week ago -- 38459).  So -- not sure what changed to allow that to work.

This is the version from the weekly builds that worked for me:

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.10
```

JohnOCFII, 2024-08-22:
Trying this again in light of this similar ticket (https://github.com/FreeCAD/FreeCAD/issues/10589) while, I was not being kicked back to the start page with a single file open, if I had more than one file open, it still failed on me with 38459, but did work correctly with the version I built locally last night with the changes from PR#15944.

So I'd say this is a valid fix that solves a few separate but related open issues.  You can close, or I can wait to close until after it is in a weekly.

closed 2024-08-25 by chennes

flaviut, 2026-09-30:
If I revert https://github.com/FreeCAD/FreeCAD/pull/15944, I can reproduce this:

```
from PySide6 import QtCore, QtWidgets

app = QtWidgets.QApplication.instance()
app.styleHints().setTabFocusBehavior(
    QtCore.Qt.TabFocusBehavior.TabFocusTextControls
)
Gui.getMainWindow().findChild(
    QtWidgets.QWidget, "NavigationIndicator"
).setFocusPolicy(QtCore.Qt.FocusPolicy.NoFocus)
```

Then:

1. Create two documents.
2. Create a Part cube in the second document.
3. Change its Length in the Data property editor.
4. Press Return.

**Expected:** The second document stays active.  
**Actual:** The first tab becomes active.

I will poke around and see if there is a cleaner fix to this that doesn't interfere with the bugfix in https://github.com/FreeCAD/FreeCAD/pull/33054

PR #15944 (2024-08-25): Gui: Workaround for #14350
On macOS, when closing a persistent property editor, Qt switches the MDIView active tab to the first one, regardless of which you are currently editing. This adds a workaround that records the active tab and explicitly sets it after closing the editor. Because there is no redraw in between these events, the switch is not visible.

Fixes #14350