FreeCAD/FreeCAD #14352: Freecad YAML parsing error all - if it is in a subfolder with a special name

Superbobo75, 2024-05-29:
If Freecad is unpacked and started from a subfolder with a special name - e.g. in Czech "_3D tiskárny" anything in Freecad stops working after starting with a YAML parsing errors:

```
YAML parsing error: 'D:/Data/_3D tiskárny/software/Freecad Dev/data/Mod/Material/Resources/Materials/Appearance/Aluminum.FCMat'
'bad file: D:/Data/_3D tiskárny/software/Freecad Dev/data/Mod/Material/Resources/Materials/Appearance/Aluminum.FCMat'
```

If I move the Freecad folder to root, it works fine.

This happens from FreeCAD_weekly-builds-37249-conda-Windows-x86_64-py311.7z

adrianinsaval, 2024-05-29:
@davesrocketshop

davesrocketshop, 2024-05-29:
I'll look into it. It's possible the string handling in the Material code has an issue. That wouldn't relate to the TopoNaming issue though. That may be an unrelated but similar issue.

PR #14356 (2024-05-29): Materials: Correct issue with UTF8 path
The YAML library is unable to open files with UTF8 paths

fixes #14352
closed 2024-05-29 by wwmayer