FreeCAD/FreeCAD #14354: [Feature Request] list of recent files - deleting selected ones

kaktusus, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Deleting selected items would be a very useful feature.
This could be accessed from the context menu when clicking on the selected item.

https://forum.freecad.org/viewtopic.php?p=761921&style=4&style=4&style=4&sid=4935b576e8e9ce9e0b520847113d7896

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.6.37
  * OpenDark 2024.3.13
  * Render 2024.1.23
  * Alternate_OpenSCAD 1.0.0
  * Plot 2022.4.17
  * kicadStepUp 10.23.2
  * dodo 1.0.1
  * Rocket 3.3.0
  * pcb 6.2023.1
  * freecad.gears 1.2.0
  * woodworking 0.21.2.33771
  * Marz 0.1.5
  * kicadStepUpMod 11.1.5
  * Movie.backup1716816115.789322 2024.2.15 (Disabled)
  * Movie 2024.2.15
  * fasteners.backup1716822852.9058104 0.5.21 (Disabled)
  * sheetmetal.backup1716822856.2948024 0.4.14 (Disabled)
  * sheetmetal 0.4.14
  * fasteners.backup1716930613.4898772 0.5.21 (Disabled)
  * fasteners 0.5.21
  * Reinforcement.backup1716930619.6547372 (Disabled)
  * Reinforcement
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12842

maxwxyz, 2024-05-29:
Also related: https://github.com/FreeCAD/FreeCAD/issues/13346

closed 2024-05-29 by maxwxyz

jorgebr, 2024-08-30:
For simple users just searching for a way how to do this, even if it's not part of the user interface:
Go to Tools > Edit Parameters ...
Then BaseApp > Preferences > RecentFiles

You can edit or delete any entries there. You must restart FreeCAD to see the changes.