FreeCAD/FreeCAD #14360: DIN 6784 or DIN ISO 13715 Edge symbols

sebaras, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Edge symbols are missing in Techdraw

### Full version info

```shell
[code]
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * CADExchanger
  * ExplodedAssembly
  * fasteners 0.4.60
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

Editable edge symbols, like writen on this webpage, are missing in Techdraw.
https://blog.nupis.de/kantensymbole-im-inventor/
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sebaras, 2024-05-29:
[image]

maxwxyz, 2024-05-29:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13572

closed 2024-05-29 by maxwxyz