FreeCAD/FreeCAD #14361: SVG Import Failure

sebaras, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

By importing an SVG File into TechDraw, the arrows went lost.

### Full version info

```shell
[code]
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * CADExchanger
  * ExplodedAssembly
  * fasteners 0.4.60
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

I created the SVG File with the newest Version of Inkscape. In Inkscape there are two arrows displayed. After importing with the general SVG Import buttom, the arrows went lost. See screenshot.
[image]
[image]

Next problem:
I put the commandline 'freecad:editable="Kantenbruch 3"' into the code after the xml:space="preserve" inhope, that the imported SVG will be editable. This wasn't working. It would be nice to have this function in future.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
Did you convert the stroke to path in Inkscape before saving the .svg?
@WandererFan should this be possible to add `freecad:editable=""` in the xml for any .svg to be able to change content or only for page templates?

sebaras, 2024-05-30:
In template it works fine with the marker. What i do not understand there is why the Text position is not in the same place by inkscape and Freecad techdraw, but this is another problem.
When you import an external svg file as content on your existing template, the marker freecad:editable="xxx" doesn't run. This would be a nice tool, then you could simply create additional svg files which are editable in the TechDraw.

HowThatWorks, 2024-05-31:
freecad:editable does work for symbols, but unlike templates you can't click on the editable text itself to change it like you can with templates. You have to select the symbol and in the property view, on the data tab there should be an Editable Texts property. You need to select that then click on the button with the ellipsis that should appear next to the value field:
[image]
If there is no editable Editable Texts property then you my need to add `xmlns:freecad="https://www.freecadweb.org/wiki/index.php?title=Svg_Namespace"` to the SVG tag.

Text misplacement can be caused by `xml:space="preserve"`. Try deleting that attribute. Non-left aligned text also has issues with proper text placement.

maxwxyz, 2024-05-31:
@sebaras does this solve the issue?

closed 2024-05-31 by maxwxyz