FreeCAD/FreeCAD #14369: New CAM Simulator drilling, single operation, and rotation issues

jkvfc, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Drilling operation looks weird. Simulation with the old simulator looks fine.
It is not possible to show only a single operation e.g. profiling. Always all operations are displayed.
Not able to rotate the part in the simulation window.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Unknown
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
```

### Subproject(s) affected?

CAM/Path

### Anything else?

![TestPlateManufacturingScreenshot](https://github.com/FreeCAD/FreeCAD/assets/136894212/6b41dcda-
[image]
6f77-4c56-a25e-f7744873f7ad)
[TestPlateManufacturing.zip](https://github.com/FreeCAD/FreeCAD/files/15484916/TestPlateManufacturing.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jkvfc, 2024-05-29:
Missing picture
[image]

maxwxyz, 2024-05-29:
@shaise FYI

shaise, 2024-05-29:
Thanks, I will take a look.

shaise, 2024-05-30:
Bug confirmed. working on it.

shaise, 2024-05-31:
There are two issues here:
1. the drilling sequence is wrong.
2. there are some opengl artifacts
I have placed a fix PR. This is for the first problem. For the second, I need to investigate.
[image]

shaise, 2024-06-01:
@jkvfc ,
I tried to compile freecad under Ubuntu but failed. Seems the instructions are outdated.
Anyway I managed to compile a standalone version on Ubuntu with QT. And I managed to find the issue. Apparently QT on Linux does not create a stencil buffer by default. 
I Have updated the PR with this fix. Should work after it is merged.

jkvfc, 2024-06-04:
shaise,
New version compiled, nice drilling operation, thanks a lot for the quick perfect fix.

My second issue above 'It is not possible to show only a single operation e.g. profiling. Always all operations are displayed.' isn't solved yet.
[image]
Regardless of whether the tick for the operation is set or not, both operations are always displayed.

shaise, 2024-06-04:
OK, I will take a look at it as well. Should be an easy fix.

jkvfc, 2024-06-05:
@shaise,
Close this issue and open a new one for the remaining problem.

closed 2024-06-05 by jkvfc