FreeCAD/FreeCAD #14373: Body problem in assembly when using shape binder

Pendelf, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you use shape binder to create one of the bodies, the first body will not be available when adding bodies to the assembly 

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.21
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.5.1
  * Pyramids-and-Polyhedrons
  * sheetmetal.backup1704622370.7006474 0.3.12
```

### Subproject(s) affected?

Assembly

### Anything else?

https://youtu.be/f-PxEIDXzF4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-29:
Does it work when you manually drag both into the assembly object in the tree view?
@PaddleStroke  FYI

Pendelf, 2024-05-30:
If you drag and drop bodies, everything works fine

PaddleStroke, 2024-05-30:
This is the same bug as https://github.com/FreeCAD/FreeCAD/pull/13780
The issue is that getRootObject is returning the root of dependency graph and not root of the tree. I made this getRootObjectsIgnoreLinks to solve the initial problem. But I have not taken into account the shape binders which are same as links.
And there are also Draft and PartDesign Clones.

We could add 

                return obj->isDerivedFrom<App::Link>() 
                    || obj->isDerivedFrom(Base::Type::fromName("PartDesign::SubShapeBinder"));

to fix the case of subShapeBinder, but that cannot solve Draft clone case for instance, as those objects are `Part::FeaturePython`. Also there might be a lot of case of other objects referencing tree root objects.

@wwmayer is there not a clean way to get the tree-root-level objects? The tree is being build so there must be a way to get that.
See Pendelf video where he shows the problem : https://youtu.be/f-PxEIDXzF4

wwmayer, 2024-05-30:
> If you use shape binder to create one of the bodies, the first body will not be available when adding bodies to the assembly

It doesn't work with the sub-shape binder (green icon) but works without problems with the shape binder (blue icon).

> is there not a clean way to get the tree-root-level objects?

The method claimChildren() of a view provider shows the referenced objects as child items in the tree view. So, all objects that are not claimed by a view provider must be top-level objects.

But does this really solve the problem? What happens if the two bodies are inside a Part container?

PaddleStroke, 2024-05-30:
So to get the tree root object you iterate through all the document objects, build the list of objects that are within at least one of the object VP claimChildren return function, then the root is the rest of the objects right?

If so I will make a 'getTreeRootObject' in Gui.

Yes it solves the problem because 'insert component' is building a tree starting from root objects. So if Part container is a root object, it will be picked up. Then it is iteratively browsed to build the tree so the bodies within are found.

PR #14388 (2024-06-17): Core: Add Gui::Document::getTreeRootObjects() Fixes #14373
Fixes #14373
@wwmayer
commit 99c30da7 (2024-06-17): Merge pull request #14388 from Ondsel-Development/core_rootObjects

Core: Add Gui::Document::getTreeRootObjects() Fixes #14373
closed 2024-06-17 by chennes