FreeCAD/FreeCAD #14375: BIM: Doors and windows are only displayed as wires

gianlucafiore, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Simply by creating doors and windows, they have no material in their frames and structures. Yes, glasses in cases containing

[image]
### This is a simple door, it doesn't affect anything else as far as I could see, it's just not displayed correctly

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-05-30:
Indeed I can confirm. This has to do with the new Materials structure I guess, but couldn't find yet what's wrong...

Roy-043, 2024-05-30:
Something has changed in the ShapeColor of objects.
Version 37543:
`obj.ViewObject.ShapeColor => (0.800000011920929, 0.800000011920929, 0.9019610285758972, 1.0)`
Version 37464:
`obj.ViewObject.ShapeColor => (0.800000011920929, 0.800000011920929, 0.9019610285758972, 0.0)`

This causes a problem here:
https://github.com/FreeCAD/FreeCAD/blob/dd7a8c3eba49e160f870536ec20af1a069a3be1e/src/Mod/BIM/ArchWindow.py#L688

Possible solutions:
A.
`        base = obj.ViewObject.ShapeColor[:3] + (obj.ViewObject.Transparency / 100, )`
B.
`        base = obj.ViewObject.ShapeColor[:3] + (0.0, )`

Solution A introduces a new behavior (use the transparency of the object), solution B restores the old behavior.

@davesrocketshop Is this indeed related to the new Materials structure?

davesrocketshop, 2024-05-31:
> Something has changed in the ShapeColor of objects. Version 37543: `obj.ViewObject.ShapeColor => (0.800000011920929, 0.800000011920929, 0.9019610285758972, 1.0)` Version 37464: [...]

Ugh. I was afraid of this. The fundamental problem is that App::Color and QColor treat the transparency value very differently and in the case of App::Color inconsistently. The old default was to ignore the transparency but that's no longer an option.

I would recommend A but B should be fine.

Roy-043, 2024-05-31:
@davesrocketshop Thanks. The Draft code will have to be updated as well then. Will do that later. I have meanwhile found an inconsistency in the 'new' behavior. See #14414.

Roy-043, 2024-06-16:
In the version mentioned below the transparencies in the DiffuseColor are ignored altogether. Meaning that both the frame and the glass are opaque. And ShapeAppearance per face does not work properly yet. Fixing this bug is not possible ATM.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

davesrocketshop, 2024-06-16:
There is a fix pending that should fix this as well #14568

maxwxyz, 2024-06-29:
@gianlucafiore @yorikvanhavre did https://github.com/FreeCAD/FreeCAD/pull/14615 fix this issue? If so I'll close it.

yorikvanhavre, 2024-07-01:
The original bug is gone. As far as I can see per-face transparency is not handled yet, but I think this issue can be closed.

closed 2024-07-01 by yorikvanhavre