FreeCAD/FreeCAD #14377: BIM: Problem creating stairs using guide lines

gianlucafiore, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The steps to follow are: 
1. Create a line with the slope of the climb 
2. Select it 
3. Use the "Stairs" command and the following error appears:

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-30:
@yorikvanhavre FYI

yorikvanhavre, 2024-05-30:
No problem here, doing this works with no error:

1. switch to front view
2. create a daft line with a slope
3. switch to top view
4. select the line
5. press the stairs button

If you can reproduce the error, could you paste here the contents of the python window? (ie. the commands that create the line and the stairs). Does it give the same problem when you paste those lines again in the python window?

gianlucafiore, 2024-06-09:
[image]

[image]
It happens to me again

test file: https://drive.google.com/file/d/11YBxG2j8rRH4YZv7gNK0Y38-0bpRjdnr/view?usp=drive_link

Roy-043, 2024-06-14:
The problem is caused by the direction of the wire. The failure occurs if the wire starts at the top of the stairs.

PR #14778 (2024-06-20): BIM: Fix base edge direction for Arch_Stairs
Fixes #14377

If the base edge of a stairs has a delta Z, the start point must have lowest Z. The `makeStraightStairsWithLanding` code fails otherwise.

Using `edge = Part.LineSegment().toShape()` is not the most efficient, but matches the pattern used elsewhere in the code.
commit 5a057391 (2024-06-20): BIM: Fix base edge direction for Arch_Stairs

Fixes #14377

If the base edge of a stairs has a delta Z, the start point must have lowest Z. The `makeStraightStairsWithLanding` code fails otherwise.

Using `edge = Part.LineSegment().toShape()` is not the most efficient, but matches the pattern used elsewhere in the code.
closed 2024-06-20 by yorikvanhavre