FreeCAD/FreeCAD #14379: Subshape binder offset creates 3D surface from planar face

northcoastam, 2024-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Adding an offset value on a subshape binder produces an unexpected result.  FreeCAD file attached.

Here's the subshape binder initially (bound to the face of the body:
[image]

And here it is after applying a 1mm offset:
[image]

### Full version info

```shell
Using 0.22 dev build, full details below:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * A2plus 0.4.60n
  * A2plus.backup1693449725.06262 (Disabled)
  * Curves 0.6.13
  * Curves.backup1693449730.5624967 0.6.13 (Disabled)
  * Defeaturing 1.2.0
  * Defeaturing.backup1693449735.7019093 (Disabled)
  * fasteners.backup1695592281.051254 0.4.63 (Disabled)
  * Manipulator 1.5.0
```

### Subproject(s) affected?

None

### Anything else?

[SubshapebinderOffset.zip](https://github.com/FreeCAD/FreeCAD/files/15491563/SubshapebinderOffset.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-30:
This seems like a regression as in 0.21.2 it looks like this:
[image]
@bgbsww maybe related to toponaming or the fix in https://github.com/FreeCAD/FreeCAD/pull/14202
@FlachyJoe FYI, you've created another issue for subshapebinder in Part Design

bgbsww, 2024-06-04:
Looking.  At a glance, it appears that we are either improperly or failing to apply a transformation to the inner wires here, thus the floating circle while the outline is perfect.  That's likely a one liner somewhere ...

bgbsww, 2024-06-05:
Honestly, I'm not sure this has ever worked correctly.  There are strange behaviors with offsets in LS3, 0.21.1, 0.21.2 and main.  And this file displays the same as the error case on 0.21.2.

There is a deep issue with subshapebinder and offsets, which is ultimately triggered because the transformation positions of the sub tshapes of the circle in the face of the pad have varying Z coordinates.

That's probably some subtlety of the face creation with the pocket, and I suspect the solution will be some way to "flatten" the face that is used to attach to.

But this does not appear to be TNP related at all, nor a regression, so I'm dropping it for now.  I may pick it up again out of curiosity, but there are TNP issues to address first.

[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * Assembly3 0.12.2
  * fcgear 1.0.0
  * DynamicData 2.54.0
  * BIM 2021.12.0
  * parts_library
[/code]

adrianinsaval, 2024-07-01:
@0penBrain do you have any opinion on this matter?

0penBrain, 2024-07-01:
@adrianinsaval Not at all and not at all. 😄

adrianinsaval, 2024-07-01:
Ups, sorry for bothering

FEA-eng, 2024-07-14:
> I'm not mistaken you had introduced offsets in shapebinders?

It was @mwganson: https://github.com/FreeCAD/FreeCAD/pull/6338

mwganson, 2024-07-14:
> > I'm not mistaken you had introduced offsets in shapebinders?
> 
> It was @mwganson: #6338

Guilty as charged.

Select the same face of the Pocket object, press Ctrl+Shift+P.  In the python console enter:

```
offset = sub.makeOffset2D(1)
Part.show(offset,"offset")
```
Code explanation: sub is the face subobject.  The function makeOffset2D() is a function provided by the Face objects to do the 2D offset on them.  (1) is to do the offset by 1mm.  Part.show() merely shows the object.

[image]

Clearly the bug is in either the OCCT 2D offsetting function (or in how it is being implemented somewhere in FreeCAD core, would need to test with Draw utility to eliminate this being a FreeCAD issue, but I like the requisite skills with Draw scripting).  It is unrelated specifically to SubShapeBinder or to TNP mitigation code.

Above image is made with OCCT 7.7.2.  I get the same result in OCCT 7.6.3 and in OCCT 7.5.2.

If I show the sub with Part.show(sub,"sub") and then try to use Part 2D offset in the GUI this seems to work, but the circle is omitted from the offset object unless I use a negative value for offset, in which case the same 3D offset effect is produced as in the above image.

chennes, 2024-07-15:
I will test against OCCT 7.8

chennes, 2024-07-19:
I tested with OCCT 7.8.0 and the nonsensical behavior of the SSB offset continues. Changing the value of the offset switches the binder all over the place. However, I get the exact same behavior in 0.21.1, so IMO this cannot be a blocker. I will leave the tag on until our next Issue Triage meeting, at which point I suggest we remove it.

mwganson, 2024-07-19:
> I tested with OCCT 7.8.0 and the nonsensical behavior of the SSB offset continues. Changing the value of the offset switches the binder all over the place. However, I get the exact same behavior in [...]

I assume you don't mean removing the offsetting feature from binders.  That would break no telling how many existing models in which it is working perfectly fine.

maxwxyz, 2024-07-19:
It's weird, now it is broken on my side in the stable version too, I wonder how I've made the screenshot above in 0.21.2...

chennes, 2024-07-21:
> I assume you don't mean removing the offsetting feature from binders. 

😱 No! Definitely not, I just mean removing the "Blocker" tag from this issue.

FEA-eng, 2025-06-30:
Still relevant

YosypC, 2025-11-27:
Could it be related? #9532

northcoastam, 2026-05-15:
Thanks. Now using OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260512 (Git shallow)
Build date: 2026/05/12 17:54:42
Build type: Release
Branch: grafted,grafted
Hash: 758570d19c79cc4cf8b1e198b5568e7263e8e6d8
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU) [ OS: English/Australia (en_AU) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/138.461/1.75
Installed mods:
  * Detessellate 1.1.0

Same results for +ve and -ve offset as Version: 1.2.0dev.20260421 (Git shallow)

[image]