FreeCAD/FreeCAD #14380: CAM LeadInOut Dressup Doesn't respect perimeter

Connor9220, 2024-05-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When applying a LeadInOut Dressup, it sometimes goes outside of the perimeter when doing multiple features in one op.

In image below, some of the squares, the tear drop, the triangle, the hexagon, all have a lead-in and lead-out that goes outside of the perimeter.   The only geometry that is not affected by this is a circle. (Not beginning and no end)  Rounded slots can be affected to, though, I wasn't able to reproduce it with this sample. 

[image]

Changing the size of the endmil can help sometimes, but, I was trying to setup a job for a plasma cutter with very small kerf and had this issue.

Here is another part with much smaller kef (plasma), lead-in and out is .100" (Yes the small holes have issues, but that's because the lead-in/out value is too large). This one displays issues with rounded slots.  

[image]

One other issue: For the Length/Radius values. They can't start with 0.0, has to be 0.1 or larger.   Though, sometimes, it does come up as a 0.0xxx if you let it take the default and the endmill size is small.  Additionally, if you don't explicitly use in or " to denote imperial, it converts your value to metric, regardless of your unit system preference. 

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * Dracula 0.0.5
  * A2plus 0.4.64a
  * Assembly3 0.12.2
  * Behave-Dark-Colors 0.1.1
  * ThreadProfile 1.89.0
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
  * Assembly4 0.50.12
  * parts_library
  * dodo 1.0.1
  * btl 0.9.9
  * fasteners 0.5.17
  * OpenDark 2024.3.13
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-30:
Could you provide a demo file as .zip?

Connor9220, 2024-05-30:
Here is a sample.  LeadInLeadOut-Test is the exact one used for the screen shot above.  ConnectorPanel-CAM-Test might be slightly different that pictured above, but still has at least two issues on the far right hand side of the part with the elongated circles. 

[LeadInLeadOut-Sample.zip](https://github.com/FreeCAD/FreeCAD/files/15503022/LeadInLeadOut-Sample.zip)

tarman3, 2025-07-01:
You are wrong
The side determined correctly, but some of start points get to corners and it looks like the side is incorrect

You can not set StartPoint for multiple profiles

tarman3, 2025-07-01:
You can check this PR #22137 to move start point

[image]

Also you select face and Process Holes and Circles
In this case LeadInOut can not correctly work

You should select each profile faces or wires

Connor9220, 2025-07-01:
> You can check this PR [#22137](https://github.com/FreeCAD/FreeCAD/pull/22137) to move start point
> 
> 
> Also you select face and Process Holes and Circles This is a complicated case in which it is mixed inside and outside direction In this case LeadInOut can not correctly work
> 
> You should select each profile faces or wires

I originally had issues with the panel plate (The one with holes and  rounded slots).  I as doing that on a plasma and expected to be able to do all of that in a single pass.  This was also filed back in v.21.2  I've not looked at this file in a while. It should have been ALL inside direction.  I had to select OUTSIDE to get it to do the INSIDE of the holes when I selected the face.  I still don't understand that behavior.  And you'll notice that I didn't select Perimeter..  Looks like to me maybe Side is relation to perimeter.. and inverted for other features?

tarman3, 2025-07-01:
For me most simple solution is to create macro which will select all closed wires from selected face
After that we can apply Profile operation and will get correct side and correct result at all

Connor9220, 2025-07-01:
Here is a screen shot where I just selected all the wires using a macro, used INSIDE and did a lead-in / out.  You'll see a few still collide with the walls. (Bottom row, the far left and far right)  It never happens with circles from what I can tell. 

[image]

Connor9220, 2025-09-27:
Still need to see if we can mitigate this.

tarman3, 2025-09-27:
To fix this problem path should start not from the corner,
but from center of long side

Connor9220, 2026-01-30:
Possibly have a fix. Keep this open.