FreeCAD/FreeCAD #14390: Part Design: can't pad a sketch on datum plane "Pad: Null shape"

janjan, 2024-05-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I use the latest weekly build (37543) and I can't extrude simple sketches if they are attached to datum planes. The padding fails with the error "Pad: Null shape"

If the same sketch is attached to the XY plane it works

[image]

Steps to reproduce:

* create new file
* create PD body
* create a Datum plane, attach it to XY, move it up by a few mm and rotate it a bit around the x axis
* create a new sketch on the datum plane and draw a circle
* pad the sketch

Here is a sample file from the screenshot:

[pad fail on datum plane.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15499837/pad.fail.on.datum.plane.FCStd.zip)

Just try to pad the sketch and you'll see the error

### Full version info

```shell
OS: KDE neon 5.27 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.5
  * Defeaturing 1.2.2
  * QuickMeasure 2022.10.28
  * fasteners 0.5.14
  * Curves 0.6.24
  * MeshRemodel 1.9.22
  * Manipulator 1.5.7
  * kicadStepUpMod 11.1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-06-01:
Must be a regression of v0.22 because with v0.21 it works.

MiqCG, 2024-06-06:
Found the same problem in revision 37543 but it is OK in 0.22 revision 35966.

MiqCG, 2024-06-06:
On the other hand, if it helps, the bug is present in Ondsel 37455 but it is not appearing in Ondsel 37179.

closed 2024-06-10 by chennes

macdroid53, 2024-06-10:
Hmm...37669

[image]

wwmayer, 2024-06-11:
The error message `Pad: Null shape` is gone but the output shape doesn't make any sense. It's a compound of two parallel faces but not a solid as expected.

Also, there is now a warning: `QObject::connect: signal not found in QListWidget`

FlachyJoe, 2024-06-11:
@macdroid53 your version doesn't include this PR
All is right with:
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Debug
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
```

@wwmayer I've informed the concerned PR author. Thanks.

macdroid53, 2024-06-11:
Yeah, after I hit send on that comment I realized I should have checked the merge. Sorry.

PR #14397 (2024-06-10): Fix can't pad a sketch on datum plane
Fix #14390