FreeCAD/FreeCAD #14395: The software crashes when I click on the import file type.

luisDias89, 2024-05-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

The problem is as follows: I start FreeCAD, create a new document, and click File -> Import. On this screen, when I click on the import file type, the software crashes and I have to end the process.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)

ondsel 2024.2 whith same problem
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-30:
I cannot reproduce in the same version and also on Win11
Does it also happen in the File>Open dialog?

```OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

luisDias89, 2024-05-30:
Okay, this problem also occurs in other windows within FreeCAD.
I have tested the OpenFile Dialog with several other software applications, including GIMP, KiCad, LibreOffice, and SolidWorks, and it is working fine. Only FreeCAD is experiencing this issue.
And the problem always occurs when I click on the file type filter.

[image]

maxwxyz, 2024-05-31:
I still cannot reproduce. Could you try this with a clean FreeCAD without the config files?

Syres916, 2024-05-31:
Please change you default lookup folder to a proper hard drive letter instead of a OneDrive and retest, something tells me that's the root cause.

luisDias89, 2024-05-31:
It's not because of OneDrive.
[image]
I use Ondsel, (same problem) and this software are instaled in right place. I also tried running it in administrator mode, and the result was the same.

luisDias89, 2024-05-31:
Finally, I found the cause of the problem: it only occurs when I have the second 4K monitor connected. If I remove the second monitor from the equation (disable it in the system), it works perfectly.

[image]

luisDias89, 2024-05-31:
This is likely a Qt issue, where it cannot render the window properly with the second monitor. If you shrink the "file dialog" and place it on the left side, the window gets rendered, but in a strange way: all the text moves to the second monitor.
[image]

Syres916, 2024-05-31:
@luisDias89 thanks for clarifying the dual monitor setup, do you have a Windows service running called Nahimic? If so, try stopping it and set to disable to stop it automatically starting on a reboot. Then retry having both monitors enabled and see if the FreeCAD import works correctly.

luisDias89, 2024-05-31:
I don't have any service called Nahimic running.

Syres916, 2024-06-01:
Thanks for checking, then I'm out of possible solutions, I agree that Qt is the likely culprit. All I would ask is could you please update the title of this issue to include 'using dual monitors' so it stands out without users having to click into the issue to find your investigative feedback, thanks.

MisterMakerNL, 2024-06-01:
I bug reported a similar issue with linkstage.
@realthunder  solution then was to use native explorer instead of windows explorer too ooen files.

Syres916, 2024-06-02:
Just incase @luisDias89 isn't aware how to trigger the native explorer, see https://wiki.freecad.org/Fine-tuning#General and the **Dialog/DontUseNativeDialog** setting

luisDias89, 2024-06-06:
@Syres916 Empty: 
[image]

luzpaz, 2024-07-29:
Any progress here ?

luisDias89, 2024-08-22:
No

luzpaz, 2024-08-22:
Is there an upstream Qt ticket we can reference and track ?

closed 2025-07-15 by github-actions[bot]