FreeCAD/FreeCAD #14400: [bug, TechDraw] Imported SVG symbols are incorrectly scaled

HowThatWorks, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I import an SVG symbol via the new TechDraw_View it's ends up being shrunk slightly. For example, this SVG:
[image]
ends up like this:
[image]
after reexporting via [wiki]TechDraw_ExportPageSVG[/wiki]. I find that I have to scale the symbol up by 8% to get it to come out the correct size.

Opening the same SVG via [wiki]TechDraw_PageTemplate[/wiki] gives the correct size.

This happens with 0.22 and with 0.21, the mis-scale factor is even the same.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.5
  * ExplodedAssembly
  * fasteners 0.4.52

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.5
  * ExplodedAssembly
  * fasteners 0.4.52
```

### Subproject(s) affected?

None

### Anything else?

Forum link: https://forum.freecad.org/viewtopic.php?t=88097

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

HowThatWorks, 2024-06-01:
I think I'm closer to understanding this issue. Looking at the SVGs exported by FreeCAD, I noticed that FreeCAD uses 10 px per mm, whilst my symbols use 1 px per mm.[1] So for an imported symbol to come out the correct size you have to scale it up by a factor of 10. I think this is where things are going wrong -- FreeCAD is somehow calculating the incorrect scaling factor. To test this I created a set of simple symbols that are just a 10x10 mm squares with increasing pixels per mm.
1 px per mm:
[image]
2 px per mm:
[image]
10 px per mm:
[image]

and one in which I set the document units to px in Inkscape (which uses the CSS standard 96 DPI):
[image]

The correct scale factors for the to apply are 10, 5, 1, and 10. But that's not what FreeCAD does:
[image]
```
   <g font-size="29.1042" transform="matrix(9.25926,0,0,9.25926,1392.41,957.407)" font-style="normal" stroke="none" fill-opacity="1" font-weight="400" font-family="MS Shell Dlg 2" fill="#000000">
    <rect x="5" y="5" width="10" height="10"/>
   </g>
   <g font-size="29.1042" transform="matrix(4.62963,0,0,4.62963,1758.46,957.407)" font-style="normal" stroke="none" fill-opacity="1" font-weight="400" font-family="MS Shell Dlg 2" fill="#000000">
    <rect x="10" y="10" width="20" height="20"/>
   </g>
   <g font-size="29.1042" transform="matrix(0.925926,0,0,0.925926,2123.92,957.407)" font-style="normal" stroke="none" fill-opacity="1" font-weight="400" font-family="MS Shell Dlg 2" fill="#000000">
    <rect x="50" y="50" width="100" height="100"/>
   </g>
   <g font-size="29.1042" transform="matrix(9.92063,0,0,9.92063,2482.49,950.794)" font-style="normal" stroke="none" fill-opacity="1" font-weight="400" font-family="MS Shell Dlg 2" fill="#000000">
    <rect x="5" y="5" width="10" height="10"/>
   </g>
```
(exported via `TechDraw_ExportPageSVG`)

You can see[2] that FreeCAD is trying to scale each square so it comes out correctly but gets the scaling factor wrong by 7.4 % for the first three symbols. For the last symbol it gets it wrong by 0.8 %, so unfortunately changing the document to use px instead of mm does not actually fix this issue.

So I don't think this is a case of FreeCAD expecting one unit and mishandling it when it gets another, I think there's a bug in whatever calculates the scaling factor applied to symbols.

Interestingly, FreeCAD has to do the same scaling when it imports a template. To make `test_symbol_export.svg` I used `A4_Landscape_blank.svg`, which uses the same 1 px per mm that I used in my symbols, so has to be scaled up by the same factor of 10. Which it is:
```
  <g transform="scale(10.000000, 10.000000)" id="A4_Landscape_blank1.svg" style="stroke: none;">
   <metadata id="metadata8">
    <rdf:RDF>
     <cc:Work rdf:about="">
      <dc:format>image/svg+xml</dc:format>
      <dc:type rdf:resource="http://purl.org/dc/dcmitype/StillImage"/>
     </cc:Work>
    </rdf:RDF>
   </metadata>
  </g>
```

This time with the `transform="scale"` attribute. I also tested this with `A4_Landscape_ISO5457_minimal.svg,` which also uses 1 px per mm, and also comes out with the correct scale factor. But! This is a trick. FreeCAD doesn't actually calculate the scaling factor to apply to the template. It always applies a factor of 10, regardless of what the actual scaling factor should be. Does this count as a bug? I don't think so, just a limitation. But it explains why opening the SVG in my original post as a template works. It just happens to be using the same px per mm as a template is supposed to.

[1] I'm getting this from the width, height and viewbox attributes in the svg element. Generally what I do is set the `viewbox` width and height (which is in px) equal to the `width` and `height` (which is in mm), like so:
```
   width="20mm"
   height="20mm"
   viewBox="0 0 20 20"
```
For SVG export FreeCAD sets the `viewbox` width and height (in px) to 10 times the `width` and `height` (in mm), like so:
```
viewBox="0 0 2970 2100" width="297mm" height="210mm"
```

[2] Note that FreeCAD applies it's scaling using the `transform=matrix(a, b, c, d, e, f)` attribute, so if `b` and `c` are zero then `a` and `d` are just the x- and y-scaling factors.

WandererFan, 2024-06-01:
From: https://forum.freecad.org/viewtopic.php?t=88097

<<Everything in the View's scene is scaled up by 10x.   This sets the display/print resolution at 10 dots/mm (254 dpi).   Without this, we get a blocky, pixelated display.  

The scaling you are experiencing is not related to this.    The effect you are seeing is from a conversion from px to mm (96 px / in => 3.78 px / mm).  Since your Svg is already in mm this conversion is not required, but is performed anyway.

The code is expecting symbols laid out in px units.  To handle other units, we would need to parse the svg file and convert any "length" attribute (like width and height in the svg element or x=, y= in other elements).   

Since a length can have a number of different units (length ::= number ("em" | "ex" | "px" | "in" | "cm" | "mm" | "pt" | "pc" | "%")?) and there are many potential length attributes, this would be a complicated bit of code.>>

HowThatWorks, 2024-07-05:
Would be it feasible to standardise everything on 1 px/mm? That way it's at least consistent with templates and it's easier to make symbols by hand.

WandererFan, 2024-12-25:
found this: https://stackoverflow.com/questions/49866474/get-svg-size-from-qsvgrenderer

apparently Qt's svg functions convert between mm and pixels using 90 dpi / 25.4 = 3.543307087 dpmm.

the rest of the world uses 96 dpi / 25.4 = 3.779527559

10mm = 35 pixels in Qt
10mm = 37 pixels in inkscape

which explains why our symbols using mm are always a bit small.

WandererFan, 2024-12-25:
in recent AppImage:
[image]

in my dev code:
[image]

WandererFan, 2024-12-25:
My dev code treats px arguments as being equivalent to the 0.1mm scene unit.   So px measurements should be 10x bigger to get the expected size.   

I could change that to make 1px == 1mm, but symbols made by people who don't know about the 10x factor will be oversized.

I would avoid the whole issue by defining my symbols using mm instead of px, but I can change the code if the consensus is that we should consider 1px == 1mm.

FC-FBXL5, 2024-12-25:
I'd second that it should be consistent with templates.

HowThatWorks, 2024-12-25:
Making symbols consistent with templates is the way to go, in my opinion (and I'm not just saying that because I've already made a bunch of symbols that use 1 px = 1 mm).

The ultimate solution would be to have FreeCAD correctly interpret the `viewBox` and `width`/`height` attributes when it imports a symbol, making the issue moot, but even then I guess it makes sense to have a standard pm/mm.

WandererFan, 2024-12-26:
Templates are laid out in mm, not px.  After this change, a symbol laid out in mm will look the same if treated as a template or a symbol.  

I don't know what a template laid out in px would look like.  Qt should interpret it correctly(?).

WandererFan, 2024-12-26:
Qt does the interpreting of height/width & viewbox.  TD just hands over the svg text to Qt.

Qt is going to interpret px values as "scene units" (0.1mm) no matter what TD does.

To make 1px = 1mm the code would need to scale the symbol by 10x.  But the code has no way of knowing if a given symbol is designed with 1 px = 1 mm or not.

Rather than making an arbitrary rule that 1 px = 1 mm, I would make a recommendation that to ensure the correct sizing of symbols, they should be laid out in mm.

FC-FBXL5, 2024-12-27:
Are there templates laid out in px? 
I thought using the viewbox and width + height would result in templates/symbols that are considered being based on mm.

WandererFan, 2024-12-29:
`Are there templates laid out in px?`

There are templates with width/height specified as mm, in and nothing (nothing implies px in svg).  mm and nothing are treated the same.  in is converted to mm first.

The TD code does not look at the viewBox (Qt's rendering engine might).  The template image is scaled to fit on a page with the width/height values.

commit a3e49023 (2025-01-06): [TD]Symbol scaling (fix #14400) (#18757)

* [TD]add legacy/new svg scaling pref

* [TD]Change Svg scaling logic (fix 14400)
closed 2025-01-06 by yorikvanhavre