FreeCAD/FreeCAD #14403: you can't constrain a wedge of a circle in scetcher

Chrishjones, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when you draw a wedge of a circle in scetcher it won't become totaly constrained and you can't pad it
[pie wedge test.txt](https://github.com/user-attachments/files/15511541/pie.wedge.test.txt)
.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * 3D_Printing_Tools
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * LCInterlocking
  * Lithophane
  * MakerWorkbench 1.0.1
  * offline-documentation
  * parts_library
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

I am not sure if I have provided enough infomation forn this problem but if you draw a circle and put un two lines from the center to the cercumfrance and drar an ark between the two lines and then deleat the circle you acn't constrain the wedge that is left.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-05-31:
Please share a (zip) file.

**It-works-here™**
[image]

Chrishjones, 2024-05-31:
Hi that way worked for me as well. Can you try drawing a wedge with 3 arches for me instead of 2 lines and an arch. Thank you for your help

FlachyJoe, 2024-05-31:
@Chrishjones It works too, even with complicated constraints.
[image]

Please give the failing file, lets don't play cat and mouse.

Chrishjones, 2024-06-01:
[three ark wedge.zip](https://github.com/user-attachments/files/15520453/three.ark.wedge.zip)
Hi again i hope this file is ok. Only a basic user of computres.  How did you constrain the arks like that in sketcher? thanks for your support.

maxwxyz, 2024-06-01:
This is done with the arc length constraint in the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Could this issue be closed then?

Chrishjones, 2024-06-01:
Yes. Thank you for your help

closed 2024-06-01 by Chrishjones