FreeCAD/FreeCAD #14404: Non-Selected faces rendered without color

kadet1090, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting faces in a PD feature that requires it all faces apart from first one and selected ones disappear.

[image]

Steps to reproduce:
1. Create new body
2. Insert Box primitive
3. Use thickness (or chamfer, or anything that allows you to pick faces)
4. Pick a wall

I did the bisect and the culprit seems to be https://github.com/FreeCAD/FreeCAD/commit/7d3f6e2ad13ae50deb6675255e0e6aec5fa5ed6a

Probably this change:
https://github.com/FreeCAD/FreeCAD/commit/7d3f6e2ad13ae50deb6675255e0e6aec5fa5ed6a#diff-9f3d005581de0d419c8c5b1d343918918c42a5cf2381f4eb6022a31890b271c6R165

### Full version info

```shell
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37065 +1 (Git)
Build type: Debug
Branch: more-at-the-end
Hash: 0590b51dabb5380ff983b4df2b7cb2cb21db2d18
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * SelectorToolbar
  * ExtremeProDark 2.9.0
  * BIM 2021.12.0
  * CurvedShapes 1.0.4
  * IconThemes
  * sheetmetal 0.3.0
  * OpenTheme 2024.5.16
  * Glass (Disabled)
  * lattice2 1.0.0
  * A2plus 0.4.60n
  * Curves 0.6.13
```

### Subproject(s) affected?

None

### Anything else?

Ping @davesrocketshop 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-05-31:
Investigating.,..

davesrocketshop, 2024-05-31:
I can reproduce the issue but backing out my changes has no effect. I'm continuing to investigate

kadet1090, 2024-05-31:
You must always try with new model as it is stored in the file - maybe that is the reason why you changes have no effect.

davesrocketshop, 2024-05-31:
> You must always try with new model as it is stored in the file - maybe that is the reason why you changes have no effect.

No. I've tested that way. The only thing my changes affect is getting a new default color value. Selection color changes are unaffected. That happens elsewhere and that's what I'm chasing down now.

davesrocketshop, 2024-06-02:
A significant merge affecting materials was just done. https://github.com/FreeCAD/FreeCAD/pull/13792

I'm not seeing the issue there so it's probably worth testing to see if this is still a problem.

maxwxyz, 2024-07-01:
@kadet1090 please check if fixed

FEA-eng, 2024-08-05:
@kadet1090 Can you recheck? I don't have this issue but I've never seen it before either.

kadet1090, 2024-08-05:
Fixed indeed, can be closed

closed 2024-08-05 by kadet1090