FreeCAD/FreeCAD #14405: Navigation: CAD Drag at Cursor broken when Transform tool is active

maxwxyz, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the default navigation style: CAD and rotation mode: drag at cursor the navigation seems broken when the Std_TransformManip tool is active.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/8643ff8e-479f-47e2-a4f5-783f7c464c09

@Rexbas I guess you are the right guy for this :)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2024-05-31:
I can reproduce, will have a look when I have time.

maxwxyz, 2025-03-09:
any news @Rexbas ?

Rexbas, 2025-03-09:
I did not do anything to fix this, I can reproduce it on 1.0 but not on the main branch. Maybe this was unknowingly resolved with changes to the transform tool? Can you still reproduce it?

OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40607 +1 (Git) Conda
Build date: 2025/03/09 14:59:45
Build type: Release
Branch: smaller-alignment-rotation
Hash: 31f8495f19cdc4b852be8db5b33780b344d30a3e
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2

maxwxyz, 2025-03-09:
I cannot reproduce either in the latest weekly.

closed 2025-03-09 by maxwxyz