FreeCAD/FreeCAD #14409: Program closes when search in file explorer, or change folder

Josfraba, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The program closes without further warning when you use the search box in the file explorer window (File Menu > Open), or if you try to change folders

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * fcgear
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-31:
Can you change filetype as this issue was presented also: https://github.com/FreeCAD/FreeCAD/issues/14395
Could you try if the issue persists in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Josfraba, 2024-05-31:
Thanks for your quick response.
 
Not exactly case #14395.
FreeCAD always closes when I try to change folders (marked in green in the example image): I press the folder icon and after a few seconds the program closes.
FreeCAD sometimes closes when using the search box (marked in red) or the address bar (blue box).
 
To open a file that is not recent, I have to resort to opening File Explorer and dragging the file into FreeCAD.

 
 
I have tried the development version that you indicate and the aforementioned error does not occur.
 
Thanks
 
Jose

Enviado el: viernes, 31 de mayo de 2024 18:27
Para: FreeCAD/FreeCAD ***@***.***>
CC: Josfraba ***@***.***>; Author ***@***.***>

maxwxyz, 2024-05-31:
Ok thanks. I'll close this issue then, we are close to feature freeze and the next version has it fixed.

closed 2024-05-31 by maxwxyz