FreeCAD/FreeCAD #14410: Measurement: Problem with the default color settings on Windows with Classic stylesheet

FEA-eng, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I downloaded the newest weekly build on a Windows computer without any other version of FreeCAD and selected the Classic stylesheet in the first start wizard. The measurement labels and lines were initially green which, I guess, is the desired style. But when I went to the Preferences, I noticed that the text and line color were white while the background color was black. I clicked OK and closed the Preferences. Now the measurement type symbol is not visible. This must be coded as the default because resetting in the Preferences also brings those colors. I just don't get why it was green initially and why it's not the actual default setting.

[image]

### Full version info

```shell
OS: Windows 10 build 19041
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-31:
The black icon also needs to be styled depending on the color/theme IMO, like @Syres916 proposed for the TechDraw preferences: https://forum.freecad.org/viewtopic.php?p=756962#p756962

maxwxyz, 2024-05-31:
@hlorus FYI
@MisterMakerNL these colors could be adjusted with the theme, it's done in OpenTheme so maybe the included ones need a defined value for this?

MisterMakerNL, 2024-05-31:
The no-stylesheet is set here:
https://github.com/FreeCAD/FreeCAD/blob/91aac05021023c946dc1545f2b13512ccf326550/src/Gui/PreferencePacks/Classic/Classic.cfg#L25
The colors for this in the other 2 theme's are from Opentheme. So it should be the same.

MisterMakerNL, 2024-05-31:
Did some tests and this is a bug, measurements doesn't use stored preference pack settings nor does it saves them if you change them.

hlorus, 2024-06-03:
> The no-stylesheet is set here:
> 
> https://github.com/FreeCAD/FreeCAD/blob/91aac05021023c946dc1545f2b13512ccf326550/src/Gui/PreferencePacks/Classic/Classic.cfg#L25
> 
> 
> The colors for this in the other 2 theme's are from Opentheme. So it should be the same.

These properties are from the previous dimensions in the Part WB. UMF uses different properties in the group "Measure":
- DefaultFontSize
- DefaultLineColor
- DefaultTextColor
- DefaultTextBackgroundColor

I'm struggling a bit to test changes in the bundled preference packs as they seem to be downloaded through the addon manager. How can they be tested locally?

I've also published a PR #14470 to paint the icons in the same color as the text. It seems like we can avoid the hack in the forum thread as we can just rely on the TextColor Property of the ViewProvider. The only limitation currently is that we assume the provided icons are drawn in black.

maxwxyz, 2024-07-01:
@hlorus any news on this

hlorus, 2024-07-01:
The icon fix just got merged, see: https://github.com/FreeCAD/FreeCAD/pull/14470

What's left is updating the themes with the properties mentioned above.

MisterMakerNL, 2024-07-01:
I am using 99% Opentheme need to do a PR to Opentheme then to here. @obelisk79 
And yes I am tagging Obelisk because I am a bit busy :)

luzpaz, 2024-08-17:
@obelisk79 care tii weigh-in ?

maxwxyz, 2025-03-09:
I'll close this as Classic is hidden on Windows on the Start page (and will hopefully be removed sometime)

closed 2025-03-09 by maxwxyz