FreeCAD/FreeCAD #14411: Measurement: Can't measure the spring wire radius

FEA-eng, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you draw a spring using a sketched circle and additive pipe, the radius of its circular ends can't be measured with the new tool, throwing this error:

`18:23:05  116.39 Selection.cpp(159): Unhandled unknown exception caught in selection observer`

[image]

[Spring.zip](https://github.com/user-attachments/files/15516305/Spring.zip)

### Full version info

```shell
OS: Windows 10 build 19041
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-31:
are you selecting the circle of the sketch or the resulting helix edge?
@hlorus FYI

FEA-eng, 2024-05-31:
> are you selecting the circle of the sketch or the resulting helix edge?

I select the edges of the 3D model. It happens for both ends (the one coming straight from the sketch and the opposite one). Maybe those aren't perfect circles anymore but the measurement tool should handle it somehow instead of throwing a generic error message.

maxwxyz, 2024-05-31:
does it work when you select the sketch circle? 
Does the QuickMeasure tool in the bottom right status bar show the correct radius when selecting the edge?

FEA-eng, 2024-05-31:
> does it work when you select the sketch circle? 

Yes, I can measure the circle itself:

[image]

> Does the QuickMeasure tool in the bottom right status bar show the correct radius when selecting the edge?

It shows the length of this edge.

maxwxyz, 2024-05-31:
So the helix seems not to produce a circular shape?
@FlachyJoe you are on the helix tool, any idea?

FEA-eng, 2024-05-31:
> So the helix seems not to produce a circular shape?

Yes, apparently. That's a separate issue though. Here I'm concerned with handling of such edges by the measurement tool.

PR #14591 (2024-06-17): Measure: Fix MeasureLength failure when measuring circles/arcs
Currently MeasureLength fails when trying to measure circles and arcs. The relevant code where the exception occurs can actually be be simplified as the rotation of the measured element is not needed to place the measure label.

Fixes #14411 
Fixes #14534
closed 2024-06-17 by WandererFan