FreeCAD/FreeCAD #14412: Preferences: Display>UI chaos

maxwxyz, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The preferences Display>UI seems very chaotic, it's a long scrollable page with checkboyes and spinboxes and it is not clear what the user changes when changing values.
Icon, Font and Itemspacing is all set to 0 by default - what is this, why is there no tooltip?

Also they should be restructured: Why is Item background color not in the Display>Color tab, as also the background color?

[image]

Maybe these could be cleaned a bit for 1.0 after feature freeze.
@FreeCAD/design-working-group FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-05-31:
The settings windows look a gazzilion time better then before, but the it definitely could use some polishing. But with the massive amount of development right now this is probably a full-time job.

FEA-eng, 2024-05-31:
With the current number of preferences, searching would make it so much better: #13538

NomAnor, 2024-05-31:
The UI preferences are still  created with python and cog from templates. The other preferences seem to be defined by hand with ui files. This should probably also addresses if the layout is changed.

jag45, 2024-06-01:
Sorry to butt in but in my experience although these settings are saved in the user.cfg file they have no effect when you close and reopen Freecad. The Icon and font size for example are reverted back to the default when displayed even though the values are still set correctly here and in the config file.

PaddleStroke, 2024-06-13:
This was discussed in https://github.com/FreeCAD/FreeCAD/issues/11004

PR #14974 (2024-07-01): Preferences: Reorganize UI and Theme pages in Display group
This changes current "Display/UI" pages to "Display/Advenced" as most of settings there were pretty technical. These settings are not removed as it is hard to decide which one should be removed and which one should stay and we are too close to UI freeze.

Settings that should be obviously kept were moved to new "Display/UI" section which now includes also options from previous "Display/Theme".

Hard to explain in words so:

Before:
[image]
[image]

After:
[image]
[image]

Fixes: #14412
commit 05aaf0ed (2024-07-01): Preferences: Reorganize UI and Theme pages in Display group

This changes current "Display/UI" pages to "Display/Advenced" as most of
settings there were pretty technical. These settings are not removed as
it is hard to decide which one should be removed and which one should
stay and we are too close to UI freeze.

Settings that should be obviously kept were moved to new "Display/UI"
section which now includes also options from previous "Display/Theme".

Fixes: #14412
closed 2024-07-01 by chennes