FreeCAD/FreeCAD #14413: FreeCAD 0.21.2 Screen Flickering

RaySQL, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am using FreeCAD and this forum for the first time.

While holding down the Alt key and moving my mouse (simple geometry) in FreeCAD my screen flickers.

I have attached a short video to demo the problem.

I am not sure why this is happening all of my other applications on my machine work fine with no screen flickering just FreeCAD.

https://youtu.be/YeS1NUETMyM

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-31:
Could you test if this also happens with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Does it happen in all workbenches or can you describe how it could be reproduced?
How is your screen setup, do you have multiple ones, different resolutions,...?

RaySQL, 2024-05-31:
Yes, I have 3 monitors, and I was able to reproduce the problem using different models.
I tried several different things however, uninstalling FreeCAD and reinstalling it resolved my problem.