FreeCAD/FreeCAD #14414: DiffuseColor and ShapeColor applied inconsistently

Roy-043, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Box.
2. Select it and invoke [Std_SendToPythonConsole](https://wiki.freecad.org/Std_SendToPythonConsole).
3. Paste `obj.ViewObject.ShapeColor` in the Python console.
4. Result (for example): `(0.8, 0.8, 0.9, 1.0)`. The last element in the tuple is `1.0`
5. Same result for `obj.ViewObject.ShapeAppearance[0].DiffuseColor`.
6. Change the DiffuseColor of the object via the PropertyView.
7. Repeat step 3 or 5.
8. Result: Last element in tuple is now `0.0`.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-31:
@davesrocketshop FYI

krushia, 2024-05-31:
Changed Transparency to "6" Property View. Notice the difference between `.ViewObject.Transparency` and `.ViewObject.ShapeAppearance[0].Transparency` below. As for the individual color types, it really doesn't make sense to use transparency in all of them anyway. (Maybe emissive transparency is for modeling cloaking fields?)

```
>>> obj.ViewObject.Transparency
6
>>> obj.ViewObject.ShapeAppearance[0].Transparency
0.05999999865889549
>>> obj.ViewObject.DiffuseColor
[(0.4779999852180481, 0.06300000101327896, 0.24300000071525574, 0.05999999865889549)]
>>> obj.ViewObject.ShapeAppearance[0].DiffuseColor
(0.4779999852180481, 0.06300000101327896, 0.24300000071525574, 0.0)
>>> obj.ViewObject.ShapeAppearance[0].AmbientColor
(0.3330000042915344, 0.3330000042915344, 0.3330000042915344, 0.0)
>>> obj.ViewObject.ShapeAppearance[0].SpecularColor
(0.5329999923706055, 0.5329999923706055, 0.5329999923706055, 0.0)
>>> obj.ViewObject.ShapeAppearance[0].EmissiveColor
(0.0, 0.0, 0.0, 0.0)
```

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23

kadet1090, 2024-05-31:
Probably related to https://github.com/FreeCAD/FreeCAD/issues/14404

davesrocketshop, 2024-06-02:
> As for the individual color types, it really doesn't make sense to use transparency in all of them anyway. (Maybe emissive transparency is for modeling cloaking fields?)

This will allow for more advanced visualizations than currently implemented in FreeCAD. You're right in terms of the previous version but not in terms of future versions.

I believe the issue in this case is in how the color chooser works. It does not allow the user to modify transparency. I need to investigate.

Roy-043, 2024-06-05:
If you change a color property by supplying a tuple with 3 values, the 4th value is automatically set to zero. This may be what the Color Chooser does. In V0.21 this is not an issue as the 4th value is always zero. In V0.22, as explained above, this is no longer the case.

The code of the Draft WB has been updated in the past to match the 4th-value-is always-zero behavior of Core. If there is a change in that behavior then I would like to adjust the Draft WB code accordingly. Will the 4th value always be 1, 0, or can the value vary depending on the transparency?

Maybe something like this should be possible?:
```
obj.ViewObject.ShapeAppearance[0].DiffuseColor = (0.8, 0.8, 0.8, None)  # None means do not change the 4th value.
```

maxwxyz, 2024-07-01:
Is this still an issue in the latest weekly?

krushia, 2024-07-02:
Does this bug include the property view's lack of transparency in the `App::PropertyColor` display, and the missing `QColorDialog::ShowAlphaChannel` option in the picker dialog?

Roy-043, 2024-07-02:
> Is this still an issue in the latest weekly?

I am afraid this bug still exists.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

Roy-043, 2024-07-02:
> property view's lack of transparency in the `App::PropertyColor` display

#15027 (fixed)
#14938 (open)

>missing `QColorDialog::ShowAlphaChannel` option in the picker dialog?

#14939 (open)

luzpaz, 2024-10-23:
@Roy-043 is this issue still in play? (I tried reproducing but got lost in the process)

Roy-043, 2024-10-23:
The original problem still exists.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

maxwxyz, 2025-03-09:
@Roy-043 is this still an issue with the latest color handling changes?

Roy-043, 2025-03-10:
This has been fixed now.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/86.4394
Installed mods:
```

closed 2025-03-10 by Roy-043