FreeCAD/FreeCAD #14415: Empty widget item warning

FEA-eng, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's a minor issue but it might be good to clean such warnings before the next release:

`"Empty widget item in QVBoxLayout 'verticalLayout_10'."`

From what I've found, `verticalLayout_10` is used in those files: https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD%20verticalLayout_10&type=code

But I don't know how to fix this.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-31:
The one that I'm aware of is caused by a Line https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/AddonManager/first_run.ui#L31

marioalexis84, 2024-06-13:
@FEA-eng do you have the steps to reproduce?

FEA-eng, 2024-06-14:
> @FEA-eng do you have the steps to reproduce?

I’ve seen it in weekly builds right after opening FreeCAD. I don’t see it in the newest weekly but I’m not sure if it’s gone for good since I’m not aware of any fixes in this regard.

FEA-eng, 2024-06-15:
Actually, it's still there in today's Windows weekly. I just have to open the preferences to see it.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.25.17
  * CurvedShapes 1.0.9
  * Curves 0.6.39
  * Defeaturing 1.2.2
  * dodo 1.0.1
  * fasteners 0.5.22
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * sheetmetal 0.4.18
```

maxwxyz, 2024-06-16:
How can I reproduce it?
I get no warning when opening FreeCAD, changing to FEM and opening preferences on Windows 11.

FEA-eng, 2024-06-16:
> How can I reproduce it? I get no warning when opening FreeCAD, changing to FEM and opening preferences on Windows 11.

I don't have to open FEM. For me, it shows right after starting FreeCAD and going to preferences.

https://github.com/FreeCAD/FreeCAD/assets/59876896/2638bac0-c7bc-41d8-ab04-04c094c4e16a

marioalexis84, 2024-06-16:
It seems that you have auto-load activated for the cfdOF and Lattice2 addons. Could you disable the auto-load of these addons to check if the warning comes from any of them?

FEA-eng, 2024-06-22:
> It seems that you have auto-load activated for the cfdOF and Lattice2 addons. Could you disable the auto-load of these addons to check if the warning comes from any of them?

I don't have auto-load enabled for any of the workbenches:

[image]

But Lattice2 has one toolbar in Part Design so maybe it's because of that?

jtbentley, 2024-07-07:
I'm getting the same warning: "Empty widget item in QVBoxLayout 'verticalLayout_10'."

It appears every time I open the preferences menu.

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64a
  * Assembly4 0.50.12
  * CurvedShapes 1.0.9
  * Curves 0.6.35
  * ExplodedAssembly
  * fasteners 0.5.17
  * Help 1.0.3
  * lattice2 1.0.0
[/code]

maxwxyz, 2024-07-07:
so you are using 0.21.2 so this bug seems to be in the stable version too.
@FEA-eng could you confirm?

FEA-eng, 2024-07-07:
> @FEA-eng could you confirm?

Yes, I have this issue in 0.21.2 stable too.

jtbentley, 2025-07-01:
I am still getting the "Empty widget item in QVBoxLayout 'verticalLayout_10'." warning when I open the preferences menu in FreeCAD version 1.0.1, but only with the Lattice2 workbench installed.

DeepSOIC, 2025-08-16:
ah, it seems to be the line. Removing it fixed it for me.

FEA-eng, 2025-10-16:
I installed Lattice2 just to check it and I don't see this warning anymore, but a lot has changed since then so it's hard to say if and how it got fixed.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
Installed mods: 
  * A2plus 0.4.68
  * lattice2 1.0.0
```