FreeCAD/FreeCAD #14418: No Sketcher Tools Active

KitchM, 2024-05-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The sketcher tools, such as copy or clone, are not on the menus nor on the toolbars.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * boltsfc 2022.11.5
  * sheetmetal 0.4.15
  * fasteners 0.5.21
  * Manipulator 1.5.7
  * BIM 2021.12.0
  * dodo 1.0.1
  * lattice2 1.0.0
  * ExplodedAssembly
  * frame 0.1.1
  * Help 1.0.3
  * freecad.gears 1.2.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-01:
Already fixed in the dev builds

closed 2024-06-01 by maxwxyz

KitchM, 2024-06-02:
And when will that be available in the next upgrade?

maxwxyz, 2024-06-02:
You can try the weekly dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Feature freeze of the next FreeCAD version 1.0 will be tomorrow, so Mid July - August is targeted for stable release.