FreeCAD/FreeCAD #14422: BIM: IfcExplorer Error when closing it

maxwxyz, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When closing the BIM IfCExplorer I get the following error message:
[image]

Steps to reproduce:
1. Call IFC Explorer
2. Do not select a file but click cancel
3. Close IFC Explorer which was opened

@yorikvanhavre FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.15
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-18:
If there is no file, there are similar errors if some of the buttons at the top of the dialog are used. The 'back' button should probably be disabled if there is no file, and also when a new file is loaded.

yorikvanhavre, 2024-06-24:
Confirmed! Working on it...

PR #14935 (2024-07-02): BIM: Fixed error when closing IfcExplorer tool
Fixes the error message that appears when closing the dialog with no file open - fixes #14422
closed 2024-07-02 by yorikvanhavre