FreeCAD/FreeCAD #14424: Multiple Monitors and undocking Tool Bars

JimmyGitYOhub, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using multiple monitors and you un-dock a tool bar, the tool tips do not display until you click on the main screen.
It's the same if you keep the toolbar on your main screen of moving it to a second monitor.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64
  * Assembly4 0.50.9
  * fasteners 0.5.13
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * sheetmetal 0.4.2
  * ThreadProfile 1.89.0
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

It doesn't matter which workbench you're using .... the issue is the same

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-01:
Could you provide some screenshots how your setup looks like and which situation is not working.

JimmyGitYOhub, 2024-06-02:
Hi Max,
I don't see how a screenshot will help because it's "live action". (forgive my potential incorrect naming conventions)
If you have access to twin monitors, undock the Combo View and drag it to the other screen. The mouse-over any toolbar in the program and you will see that no tool tip will display.

Also, I loaded 0.21.2 and discovered that it does the same thing ....

Thanks,
Jim Penn

xtemp09, 2024-06-03:
This a standard _Qt_ behavior. To show tooltips of inactive windows, one needs to set the `Qt::WA_AlwaysShowToolTips` window flag, like this:[^1][^2]

```cpp
setAttribute(Qt::WA_AlwaysShowToolTips, true);
``` 

[^1]: https://doc.qt.io/qt-6/qwidget.html#setAttribute
[^2]: https://doc.qt.io/qt-6/qwidget.html#toolTip-prop

dzid26, 2025-01-20:
https://github.com/FreeCAD/FreeCAD/pull/19154 doesn't seem to fix tooltips for undocked tool bars. 

[image]

Also btw, "Multiple Monitors" is irrelevant.

closed 2025-07-15 by github-actions[bot]