FreeCAD/FreeCAD #14426: BIM: warnings when creating a NativeIFC project

furgo16, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

## Steps to reproduce

1. Start FreeCAD
2. Activate the BIM workbench
3. Click on `BIM_Project` command 
4. The following warnings are shown:

```
14:29:27  WARNING. `product` argument is deprecated for API method "aggregate.unassign_object" and should be replaced with `products`.
14:29:27  WARNING. `product` argument is deprecated for API method "aggregate.assign_object" and should be replaced with `products`.
14:29:27  WARNING. `product` argument is deprecated for API method "aggregate.unassign_object" and should be replaced with `products`.
14:29:27  WARNING. `product` argument is deprecated for API method "aggregate.assign_object" and should be replaced with `products`.
14:29:27  WARNING. `product` argument is deprecated for API method "aggregate.unassign_object" and should be replaced with `products`.
14:29:27  WARNING. `product` argument is deprecated for API method "aggregate.assign_object" and should be replaced with `products`.
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) Snap 1037
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * OpenTheme.backup1716114997.8759558 2024.5.16 (Disabled)
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
@yorikvanhavre FYI

kaiwas, 2024-06-04:
I also get an error, but a different one.
`Level: get_decomposition() got an unexpected keyword argument 'is_recursive'`
I don't know what this is connected with.
I tried to perform the same actions on a freshly installed system and this error does not reproduce there.

https://github.com/FreeCAD/FreeCAD/assets/32273833/d892f94f-5d8d-4c33-9fc1-013fbbc7f5cb

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) AppImage
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.5.16
  * DynamicData 2.62.0
  * FreeCAD-NativeIFC 2023.10.27
  * FreeCAD_Info
  * boltsfc 2022.11.5
  * Curves 0.6.37
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.16
  * PitchedRoof
  * Reinforcement
```

yorikvanhavre, 2024-06-06:
I think this could be closed in favor of #14428 ?

closed 2024-06-06 by maxwxyz