FreeCAD/FreeCAD #14427: BIM: Native IFC projects use the same "IFC" icon for all objects

furgo16, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a NativeIFC project, all items in the tree lose their original icons (building, wall, etc) and are replaced by the same "IFC package" icon. Even with small projects, that makes it visually difficult to navigate through elements while creating a model. There might be a specific motivation, but even to someone who has been using the BIM/Arch workbenches for a while, it feels like a regression/feature loss.

Before:
[image]

After:
[image]

The "After" capture was made with the attached file: open the file and manually drag and drop all tree entities to be under `Project`.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) Snap 1037
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * OpenTheme.backup1716114997.8759558 2024.5.16 (Disabled)
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

BIM/Arch

### Anything else?

- [File to reproduce](https://github.com/user-attachments/files/15521829/IFC_icons.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-01:
I guess @marcuspollio is tackling this in https://github.com/FreeCAD/FreeCAD/issues/13895
Please let me know if this is true, then we can close this issue.

furgo16, 2024-06-01:
Thanks.

1. Regarding redesign, I cannot see `IfcProject`, `IfcSite`, `IfcBuilding` and `IfcBuildingStorey` in the task list at #13895  
2. Regarding the actual issue, the Wall icon is also replaced, which I found confusing. Will there be two sets of icons for each entity in the future? E.g. `Wall` and `ifcWall`?

yorikvanhavre, 2024-06-06:
Indeed this is provisory, and will be taken care of as part of #13895

> Regarding redesign, I cannot see IfcProject, IfcSite, IfcBuilding and IfcBuildingStorey in the task list at https://github.com/FreeCAD/FreeCAD/issues/13895

They exist already

> Regarding the actual issue, the Wall icon is also replaced, which I found confusing. Will there be two sets of icons for each entity in the future? E.g. Wall and ifcWall?

The wall icon hasn't been replaced, AFAIK... But there have been some icon updates recently.. maybe that's what you're referring to?

furgo16, 2024-06-06:
> They exist already

Thanks! Is there a branch or PR to link to this issue and see those icons? That would probably answer the second part of the question as well.

yorikvanhavre, 2024-06-20:
I'm afraid they have alll been merged already... But by looking at the code history you should be able to see if when it has last been changed

furgo16, 2024-06-20:
I think we're talking about two different things. In https://github.com/FreeCAD/FreeCAD/issues/14427#issuecomment-2143456484 my question was essentially: why do the icons (`IfcProject`, `IfcSite`, `IfcBuilding` and `IfcBuildingStorey`) still look like the description's screenshot, if they have been updated already?

In any case, before it gets more confusing (sorry!), no need to reply. I'll wait for @marcuspollio to submit the PR(s) for https://github.com/FreeCAD/FreeCAD/issues/13895 and then I'll go back to the present issue. Thanks!

marcuspollio, 2024-06-20:
Hi @furgo16 @yorikvanhavre @maxwxyz 

For info, the linked task #13895 is just a list to gather work related to the icons (design, new icons needed for future features, etc). 

IMHO, this present issue (14427) is of a code one.
The icons to be used (Arch_Building_Tree, Arch_Site_Tree, Arch_Floor_Tree) are still in the code base.
In fact, using the Toolbar to create Arch_Site, Arch_Building and Arch_Level NOT in an IFC_Project displays the right icons in the Tree (like on OP screenshot or below):

[image]

I've looked briefly in the code but I'm not familiar with it at all... there's in ifc_viewproviders.pv L:77 the mention of this IFC_object icon, but have no clue if it's a good lead.
I shall gladly leave the resolution of this question to more expert hands =D

furgo16, 2024-06-21:
Thank you for the clarification! @maxwxyz, given the input, could you remove the https://github.com/FreeCAD/FreeCAD/labels/Missing%3A%20feedback tag? Thanks!

furgo16, 2024-06-24:
Thanks for the clarification. On the camp of people who might not be interested in IFC, but would still like to occasionally export their Arch objects models as IFC for

yorikvanhavre, 2024-06-24:
> Thanks for the clarification. On the camp of people who might not be interested in IFC, but would still like to occasionally export their Arch objects models as IFC for sharing with those who don't use FreeCAD: will it still be possible to export valid IFC files?

Yes! It uses the same system. The general idea here is not to start using IFC right from the scratch. It would be too restrictive. But, imagine you are working with others (engineers, other architects...). So you model in FreeCAD, and you export an IFC file from time to time. Now imagine the project has been approved, and nobody should change the project anymore (hopefully). However, now you want to track each and every change precisely. So that's the moment you switch to NativeIFC. You could reimport the IFC file you just exported, but you can also simply convert your model to NativeIFC. When that is done, from now on, the IFC file is the "master" file. Any change is recorded in it. It's not as free-form and convenient as before (at least for now), but on the other hand you're sure your changes are limited and recorded in an identificable way.

For me this corresponds to different stages of a project. Of course the long-term idea is to be able to unify it all, and have everything with both the convenience of current arch tools and the reliability of NativeIFC

furgo16, 2024-06-24:
I'm starting to understand the workflow better. These clarifications are extremely useful, thanks.

> So you model in FreeCAD, and you export an IFC file from time to time.

I might not have phrased my initial question correctly (apologies): given your example, at the stage where the project has not yet been approved and we're NOT using NativeIFC, would the exported files' format be valid IFC, even if they do not have a `Project` container?

I realize this is more of an IFC standards question. I'm trying to understand whether the exported files with FreeCAD 1.0 would be usable by folks using other software.

yorikvanhavre, 2024-06-24:
Yes. If you didn't uncheck any of the IFC options in the preferences, the exporter will add a default project, a default site and a default building if those are missing in your model.

marcuspollio, 2024-06-25:
Sorry my reply here is off-topic relative to Tree View icons issue...

> For me this corresponds to different stages of a project. Of course the long-term idea is to be able to unify it all, and have everything with both the convenience of current arch tools and the reliability of NativeIFC

IMHO the trickiest aspect here is to allow a high degree of flexibility in any scenario, context and workflow : whether you work strictly according to the IFC schema or more "freely" with just a few Arch/BIM/Draft/Part tools, and everything in between.

I wouldn't make any distinction based on the project phases, as this varies greatly in practice.
More and more, as early as the feasibility studies, depending on the client and partners, LOD 100 BIM models are required. On the other hand, for certain projects, no files other than PDFs will be exchanged until the very end...

Of course, this whole Arch/BIM/NativeIFC is a work in progress and will require adjustments over time. The good thing at least now is that we can focus resources in one place =D

yorikvanhavre, 2024-06-26:
> The good thing at least now is that we can focus resources in one place =D

Yes, definitely. I'm really happy to see the amount of collaboration and input is already rising a lot. We should have done that a long time ago :sweat_smile:

furgo16, 2024-06-26:
Ok, I think with https://github.com/FreeCAD/FreeCAD/pull/14959 I finally understand it, and I can see the IFC-specific icons shipped separately in their `Resources/icons/IFC` folder.

PR #14959 (2024-11-15): BIM: Specific IFC icons
NativeIFC objects will now look for an icon matching their class before using the default one.

Fixes #14427

This should be merged after 1.0
commit ea26350f (2024-11-15): BIM: Specific IFC icons - fixes #14427
closed 2024-11-15 by maxwxyz