FreeCAD/FreeCAD #14428: BIM: NativeIFC project creation workflow needs refinement

furgo16, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The process to create a native IFC project fees a bit confusing. Some points that contribute to that:

1. When creating a BIM FreeCAD document, and inserting a NativeIFC project via the `BIM_Project` command, the user is asked to save twice: once as a FreeCAD project and once as an IFC project.
1.  When inserting a NativeIFC project via the `BIM_Project` command, the element in the tree is named `_ifcProject`. That sounds like a code/internal name and feels like an unfinished product. Could it not be simply named `Project`?
1. Original "BIM_Project" command icon replaced with "IFC document" icon, but Arch_Site, Arch_Building, Arch_Level commands retain their original icons and can be used outside of an IFC project.
1. Uniform IFC icon vs. distinctive FreeCAD icons (https://github.com/FreeCAD/FreeCAD/issues/14427)

I understand that the process probably just needs refinement after the BIM/Arch merge, but I think users would appreciate some guidance on the intended workflow.

Happy to file a separate issue for each point if it makes sense, but I'd like to understand that workflow to better be able to file them if needed.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) Snap 1037
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * OpenTheme.backup1716114997.8759558 2024.5.16 (Disabled)
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-01:
@yorikvanhavre FYI

yorikvanhavre, 2024-06-06:
Definitely there are several problems with NativeIFC. The main source of bugs is the NativeIFC document observer that wrongly assumes some choices. I'll work on that ASAP (I'm buried under a mountain of work for the next 2 weeks :/ ).

Basically NativeIFC objects and old Arch objects are different things completely. NativeIFC objects don't get their geometry calculated from parameters, they get their geometry from the IFC file. When working in NativeIFC mode, arch (or any other) are converted to IFC on the fly after they're created. So the same button still creates one or the other, but the object is not the same.

I agree this is confusing and needs some clarification. There is still a LOT of work needed on NativeIFC.

Basically the bottom line is: It should be possible to disable NativeIFC entirely and for people to not care about it. That's something I'd like to have for 1.0. At the moment, the IFC button in the status bar is what controls it, but there are still many bugs, and it might not be enough.

> If the user has a model saved in two formats at once, which one should they be working on every time they want to modify that model?

In native IFC mode, the IFC file is the single source of truth. The FreeCAD file is only a container. The idea of the NativeIFC mode is that you don't even need to save FreeCAD files anymore. But as said above, it's still buggy

> When inserting a NativeIFC project via the BIM_Project command, the element in the tree is named _ifcProject. That sounds like a code/internal name and feels like an unfinished product. Could it not be simply named Project?

This was to indicate that the object has no name, and received a "generated" name... 

More generally, the observer also needs to be fixed because apparently it is not correctly removed when the BIM WB exits.

yorikvanhavre, 2024-06-25:
Trying to summarize here what needs to be done:

* Locked mode is always off when starting a new file, unless Preferences -> BIM -> NativeIFC -> Lock new files is on. Difficulty: No way to detect accurately when a new file is created (or to distinguish from opening a file)
* Locked mode is always on when opening an IFC file
* Locked mode does not change when importing an IFC file
* What to do when opening a FreeCAD file containing an IFC project?

furgo16, 2024-06-26:
Thanks. I'm not sure if the concept is new, so apologies if the question is obvious: could you expand on what "Locked mode" is?

yorikvanhavre, 2024-06-26:
I just added to https://wiki.freecad.org/BIM_Workbench#General_concepts ;)

yorikvanhavre, 2024-07-03:
The above commit should fix the remaining warnings, popups, etc. caused by nativeIFC.

There is still a lot of work needed in nativeIFC, but at least now it's out of the way, and people who don't need it will not have to deal with it.

furgo16, 2024-07-03:
Reading https://wiki.freecad.org/BIM_Workbench#Working_with_IFC, a few things:

1. In Locked mode, how is the IFC file saved? Via the regular Save command?
2. If you start FreeCAD, switch to the BIM Workbench and load an IFC file... In which mode (locked, unlocked) will be the BIM Workbench in? Or do you need a "container" .FcStd file (which always links to the .ifc file) that needs to be loaded, to be able to autodetect the mode?
3. The first paragraph on that section ends up with a `Read more on [NativeIFC]` section with an empty link.

yorikvanhavre, 2024-07-03:
Yes, I didn't finish writing ;) More to come!

PR #15142 (2024-07-10): BIM: Fixed remaining warnings and popups
Fixes the remaining reported issues caused by the NativeIFC merge
Fixes #14428
commit 5ae220b9 (2024-07-10): BIM: Fixed remaining warnings and popups - fixes #14428
closed 2024-07-10 by yorikvanhavre