FreeCAD/FreeCAD #14431: [TechDraw][UX]Dimensions that attach magnetically instead of Align Dimensions tools

Reqrefusion, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is always a problem to place the dimensions added in FreeCad TechDraw. Align Dimensions tools have been added to make this easier. In my opinion, it is one of the well-thought-out features in FreeCad in terms of being user-friendly. The fact that these tolls are divided into three is perhaps their most important flaw. Their behavior is to align and center the measurements.

I think these tools would be easier to use if the alignment behavior was replaced with a dimension that snaps to another dimension if there is one on the side and enters a parent relationship. In this way, alignment can be easily done when relocating the measurement without using a special tool for alignment. It also provides great convenience to move aligned dimensions collectively.

Of course, there must be a tool for centering the measure, but it would be much better if it were a single tool.

### Full version info

```shell
All Versions
All Operating Systems
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
If this is an issue about snapping, this is already implemented with https://github.com/FreeCAD/FreeCAD/pull/13659

maxwxyz, 2024-06-02:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11636

closed 2024-06-02 by maxwxyz