FreeCAD/FreeCAD #14432: UI: Object Selection controls are inconsistent

kadet1090, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A lot of features selecting some existing elements as their parameters, for example in PD:
 - PD: Thickness require selecting faces
 - PD: Chamfers and Fillets require selecting edges or faces
 - PD: Transforms require selecting features
 - PD: Pad can require selecting face in UpToFace mode
 - PD: Various features require selection of axis 
 - Part: Chamfers and Fillets require selecting edges or faces
 - etc

This is a very common task, yet it is implemented in almost every scenario independently which results in major inconsistencies:
[image]

We need to create Qt control that can be reused for such tasks providing consistent user experience across different workbenches and features.

Such control should at least:
- allow selecting any type of object / sub-object (features, bodies, solids, faces, edges, vertices etc.), including multiple types at once
- support multi-selection
- ensure that selected items are highlighted in the 3D View
- support extending to provide more inputs (for example per edge chamfer radius settings)

Example for other CAD programs:
[image]
[image]
[image]

### Full version info

```shell
non relevant
```

### Subproject(s) affected?

Core

### Anything else?

@Design

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
Another example is FEM, where sometimes you need a preselection and then click "add feature" to be able to add a selection, sometimes you need to click "add feature" and then select one (or multiple) features and in other dialog you need to click "add feature" after every selection in the 3D view again.

kaiwas, 2025-02-18:
I would like to add that sometimes you just need to select the entire object.
There is some confusion now when selecting an object. The object is selected in the tree view and, for example, the face that was clicked. This is not a problem for those who are used to the current situation, but for new users it is a little unclear.

I agree that there are a lot of problems with the selection. Everything is very unstable.

kadet1090, 2025-02-18:
My current mockup for the possible UI looks like this:
[image]

I've decided on vertical layout for parameters in case of multiple selection as this is the most generic case. We have limited horizontal space and we names of selected geometries can be pretty lengthy. Vertical layout will not be as information dense as table but it will provide enough space for data input, label and multiple parameters - something that could be hard to achieve with horizontal layouts.

maxwxyz, 2025-02-19:
In some tools like fillet, we also have the select button switch to preview. Is this intended for this UI too? Or will the preview just always be visible when the dialog is active, and when clicking select it will show it without preview? 
This is of course not necessary, if the new preview PR is merged as preview would always be visible (maybe it would need a button to show the real preview without the previous feature overlaid)?

kadet1090, 2025-02-19:
Preview should not be part of selection. I hope that new preview system will make that not needed, but in other cases preview should be always on (like it is) and should be disabled temporarily when "select" is clicked and user does selection in 3D view.

MisterMakerNL, 2025-02-25:
You got my support on this. It's always a bit selecting stuff in Freecad.

Pendelf, 2026-09-14:
https://github.com/user-attachments/assets/d5c5f01d-ce32-402f-a2e7-d6e2e931c883

It’s really very unintuitive to choose