FreeCAD/FreeCAD #14436: PartDesign: No planar face in AttachEngine3D::calculateAttachedPlacement

7th-fret, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

In console i get this:

```
<PropertyLinks> PropertyLinks.cpp(453): Unnamed#Sketch001.Support missing element reference Unnamed#Pad ;g1841;SKT;:H1160,E;FAC;:H1160:4,F;:G0;XTR;:H1160:8,F.Face6
Sketch001: No planar face in AttachEngine3D::calculateAttachedPlacement()!
``` 

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.12.3, Qt 5.15.14, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
@bgbsww FYI

maxwxyz, 2024-06-02:
Could this be resolved with https://github.com/FreeCAD/FreeCAD/pull/14429 ?

7th-fret, 2024-06-02:
> Could this already be resolved with #14429 @FlachyJoe is this the same issue?

Just tested with current main branch (37578), the problem remains.

Syres916, 2024-06-04:
@7th-fret looks like we've got progress, works error free with:

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37649 (Git)
Build type: Release
Branch: main
Hash: 681e8c9d2f95d4341e48cec629267182b2f7e586
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```
[image]

7th-fret, 2024-06-04:
> @7th-fret looks like we've got progress, works error free with:

@Syres916    I can't confirm that it works with latest main branch (Rev. Num. 37649)

The only way i found to make it work is:

1. add sketch with rectangle, pad it
2. on top face of pad create new sketch and make a new rectangle and pad it
3. edit first sketch and add an arc on bottom line, trim, exit sketch (recompute fails)
4. edit 1st sketch again, revert the changes made in previous step, exit sketch (the body recomputes OK)
5. edit again and make the same edit as of step 3, exit sketch (recompute works OK) 

Tested with revision numbers  37574  and  37649

[image]

```
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37649 (Git)
Build type: Release
Branch: main
Hash: 681e8c9d2f95d4341e48cec629267182b2f7e586
Python 3.12.3, Qt 5.15.14, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

Syres916, 2024-06-04:
@7th-fret thanks for the feedback, I can now create it error free from scratch _and_ I can create it with errors like you so this is a prime example where the developers need a Unit Test written in Python to test their changes against.

I'll try and build it tomorrow. I think the fact you create the sketches in positive X and positive Y whereas I create rectangles with the origin in the middle _maybe_ a factor but of course it shouldn't be.

Syres916, 2024-06-05:
@bgbsww I've created two new tests https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:TNP_Test_No_planar_face_in_AttachEngine3D_calculateAttachedPlacement I'll be perfectly honest, this issue has been labelled with TNP but I'm still to be convinced. You're probably going to chuckle at this suggestion but could the way the arc is created (clockwise versus counter-clockwise) be the root cause? The fact that one test always passes and the other always fails is just bizarre.

I'm now looking at making a further test but this one I'm happy is 100% TNP and will report back when it's complete.

Syres916, 2024-06-05:
@bgbsww added _testPadChange_UpToFirst_to_Dimension_ to the same branch, note the Assumption comment regarding the PropertyLinks warning.

bgbsww, 2024-06-05:
Not laughing at all at the arc direction idea; I think I've seen a little of that.  OCCT has a concept of curve orientation - Forward or Reversed, and if we are failing to check or correct that somewhere, then it would work when drawn one way but not the other.  The first two tests look solid to me, although I'd be surprised if the quadrant affected the results.  Good to prove that though.  The last test I'm as suspicious as you are of the warning.  None of the TNP warnings were placed with the intention of them firing, so it is a potential clue to a place where something is going wrong.  Thanks for writing those, it is helpful.

I suspect this may be a place where TNP changes did break it, but it isn't actually a TNP problem.

7th-fret, 2024-06-06:
@Syres916 @bgbsww 

Doing some testing to pinpoint when the problem occurs, I came up with the following.

1. The shape of the original sketch (first) didn't matter. A rectangular, polyline, polygon... designed either from sketcher's ready-available shapes or created manually with lines one by one.

2.  It didn't matter the shape used for the change. Can be an arc, polylines, part of polygon... something after all that will change the original one line into many. In case of arc, it didn't matter the way the arc was created (clockwise or counter-clockwise)

3. It didn't matter where the sketch was drew,  positive X, positive Y , negative X, negative Y, at origin...

In all tests i made, the recompute failed ONLY when the first line of sketch (Edge1) was changed

bgbsww, 2024-06-06:
I can definitely cause this to happen.  I'm currently suspicious that is about changing the number of outline elements in the sketch, which then changes the number of faces in the first pad - and that causes the attachment reference to either a face that doesn't exist anymore or a face that has a new history to fail.

So this bug is about failing to map the old attachment to a new one when the underlying face has changed, and may or may not be pad specific.

CalligaroV, 2024-06-06:
Hi Everyone.

Probably I found a fix that should close this issue.
If reproduce the steps of this issue with this modification the reference for the second sketch is updated correctly and the final shape contains the added edges/faces.

The modifications are in #14536, available for tests.

p.s.: @Syres916, I added also the tests you developed in https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:TNP_Test_No_planar_face_in_AttachEngine3D_calculateAttachedPlacement.
Performing them without my modifications triggers the `AttachEngine3D` error in the test `testRectanglewithArcChangeinGlobalUpperRight`, while performing them with my modifications doesn't trigger the error.

Thanks for those tests!

7th-fret, 2024-06-07:
Αll tests which i had done and failed, with PR [#14536](https://github.com/FreeCAD/FreeCAD/pull/14536) was successful

CalligaroV, 2024-06-07:
> Αll tests which i had done and failed, with PR [#14536](https://github.com/FreeCAD/FreeCAD/pull/14536) was successful

Glad to hear that!

Thanks for testing it!

macdroid53, 2024-06-10:
This is still present in:
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git)
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * lattice2 1.0.0
  * Curves 0.6.38
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.5.16
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.18
  * Silk 0.1.5
  * kicadStepUpMod 11.1.8
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.22
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9

CalligaroV, 2024-06-10:
Hi @macdroid53.

I confirm, the issue is present also with #14536, while on LS3 it works.
However in main, both with or without #14536, if I follow these steps:
1. reproduce your case
2. restore the original rectangle
3. apply the polyline
the issue isn't present anymore

Looking with more attention I see that after deleting the vertical edge the first time (step 1) there are no coincidence constrains between the ends of the polyline and horizontal edges, while after redrawing it (steps 2 and 3) the constraints are still present.

There should be something happening in LS3 that updates also the constraints after deleting a geometrical object related to those constrains

depthoffocus, 2024-06-13:
I'm not sure if I should add this here or on a new ticket, but I get a similar issue in 37709 (and in the previous weekly) with a revolve.

It occurred when trying to make a typical pulley wheel with a cutaway, but I've reduced it to this simple test case.

Observations: 

- it breaks this way if the sloped edge is the second or third edge added
- if the sketch is started on the sloped edge, the pocket moves to the bottom face
- if the sloped edge is the last face, it doesn't break.

[image]

PR #14536 (2024-06-26): Part/Toponaming: makeElementWires - import from LS3 fork
Change the definitions of the `makeElementWires` methods, aligning them with LS3 fork.

For reference:
https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Mod/Part/App/TopoShapeEx.cpp#L3004
https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Mod/Part/App/TopoShapeEx.cpp#L3181

In my tests this fixes #14436
closed 2024-06-26 by chennes