FreeCAD/FreeCAD #14438: Sketcher : OVP doesn't use same color after saving the preferences

0penBrain, 2024-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
 * Start FC with no user.cfg/system.cfg
 * New document, new sketch, draw a rectangle
[image]
 * Open preferences and just save them without changing anything
 * Draw a rectangle again => :warning:
[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37574 (Git)
Build type: Debug
Branch: master
Hash: 958d83ed06ca32294227b4e1270cf41c838962e5
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
@PaddleStroke FYI
@obelisk79 how it is done with your theme, it seems to work with OpenTheme

PaddleStroke, 2024-06-04:
I can reproduce.
The issue is not about OVP, OVP are reusing the colors or constraints and deactivated constraint. Same issue happen with deactivated constraints. I'll fix it

closed 2024-06-10 by chennes