FreeCAD/FreeCAD #14440: [Assemlby] : Double-clicking on an assembly should switch the workbench to the assembly workbench

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For better interaction during development, a double-click on the assembly should switch the workbench to the assembly workbench. This is my suggestion.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2024-06-02:
It already does, but not when the Assembly object it active in the tree view.

Steps to reproduce:
- Right click assembly, uncheck `Active object`
- Switch to another workbench
- Double click assembly, and you are back in the Assembly workbench.

Another way of going back to Assembly workbench and the assembly object is active, is to just double click twice on the assemvbly object.

closed 2024-06-02 by maxwxyz

luisDias89, 2024-06-03:
@howie-j Yes, I managed to reproduce it. But I found it so unintuitive that it's better to remove the functionality. Nobody is going to be opening and closing and changing workbenches for this to work