FreeCAD/FreeCAD #14441: [Assembly] : It is not possible in this version to directly create an assembly as a child of the project root.

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is not possible in this version to directly create an assembly as a child of the project root.

https://github.com/FreeCAD/FreeCAD/assets/48370099/be5e6ef7-2b94-4502-9b1f-7a1fd7496e32

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
Currently, only one assembly per root object is allowed as described also in the tooltip.

luisDias89, 2024-06-02:
@maxwxyz 
Okay, so change this topic to BUG, because if I drag the assembly, it allows adding two assemblies per root object.
Watch the video I sent carefully. I can do it, but only after the assembly is created; I can drag it with the mouse.

PaddleStroke, 2024-06-04:
> only one assembly per root object

This rule has been enforced not for technical reasons. You can have several root assemblies without any issues.
I added this limitation because a lot of people asked for it and keep asking to limit to one assembly per doc. I was against this but complied as it seemed to be what most wanted.

But yes the tool is inactive if you have already a root assembly, but nothing prevents you from drag/drop a sub assembly to root level. I'm not interested to solve this. Because in the first place my wish is to scrap this limitation altogether. It's useless and I don't want to spend more time working on it / maintaining it.

Be-Cool-Man, 2024-06-05:
It would be interesting to know why so many people want this restriction.

luisDias89, 2024-06-16:
> It would be interesting to know why so many people want this restriction.

> It would be interesting to know why so many people want this restriction.

Maybe as a matter of discipline . . .