FreeCAD/FreeCAD #14442: [Assembly] : Mouse movement is blocked, and the transformer has a bug

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am testing the new assembly, and I am finding many bugs. I will report them one by one to see if they can be resolved.

For this one, I'm not sure what happened. It is no longer possible to move the parts directly with the mouse, and the transform cursor is on the wrong object but moves the correct one.

https://github.com/FreeCAD/FreeCAD/assets/48370099/a8f1cd32-54c3-4d99-a15a-d65c58828b3d

[Teste1.zip](https://github.com/user-attachments/files/15523862/Teste1.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
I can move it in the provided file. Maybe deactivate and activate the assembly object with right click and see if this resolves the issue.

PaddleStroke, 2024-06-05:
I'm not sure what is happening in the video.
If you use the transform tool, you are going to deactivate the assembly. Because the transform tool is taking the edit.

The 2 ways to move an object in assembly is either : 
- click it in the view and drag it.
- double click an object (or select in the tree) and a dragger will appear (same as the tranform dragger)

Zikoel, 2024-10-15:
Same problem here with a most recent version

https://github.com/user-attachments/assets/e088131a-0e94-4c80-9a17-d69513baf796

[example.zip](https://github.com/user-attachments/files/17384708/example.zip)

### Full version info
```
OS: Ubuntu 24.04.1 LTS (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * fasteners 0.5.29
```

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

closed 2025-09-27 by luisDias89