FreeCAD/FreeCAD #14443: [Assembly] : Circle and 1 point assembly broken

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The assembly of a point and a circle is exhibiting unexpected behavior. I have posted a video in another software with an example.

Freecad problem:
https://github.com/FreeCAD/FreeCAD/assets/48370099/6ebd9365-0b93-4db3-a8e9-5af0de4262dd

Exemple off good work:
https://github.com/FreeCAD/FreeCAD/assets/48370099/e109c01b-32ad-4454-ab91-d064c5a2eb41

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
@PaddleStroke FYI.
I'm not sure if a distance constraint between a point and line is currently supported?

PaddleStroke, 2024-06-03:
It should be I think. I will need to check this out. Feel free to assign me on issue when you think it makes sense @maxwxyz . This way I can keep track of them.

luisDias89, 2024-06-16:
> @luisDias89 in the future, best bundle the file for the bug reproduction. Makes life easier to debug. I can reproduce the error logs that you get when creating the joint :
> 
> > File "C:\Users\Sky_l\Desktop\FC build\Mod\Assembly\JointObject.py", line 502, in setJointConnectors
> > solveIfAllowed(assembly, True)
> > File "C:\Users\Sky_l\Desktop\FC build\Mod\Assembly\JointObject.py", line 145, in solveIfAllowed
> > assembly.solve(storePrev)
> > <class 'IndexError'>: Shape index 4 out of bound 2
> 
> This error message apparantly comes from `TopoShape::getSubTopoShape` I will investigate.

If you create any more issues, I will include the file. :)

aiksiongkoh, 2024-07-23:
I will implement the Point in Curve joint next. Thanks.