FreeCAD/FreeCAD #14444: [Assembly] : The object that is grounded starts moving. BUG.

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/48370099/fa92a66f-1a4b-483b-a1d5-d716e892c720

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
@PaddleStroke FYI
This seems to be another bug with the distance constraint.
Should I assign all confirmed issues to you?

PaddleStroke, 2024-06-03:
Yes please.

PaddleStroke, 2024-06-11:
Grounded parts should not move anymore after https://github.com/FreeCAD/FreeCAD/pull/14549.
If a grounded part is moved by a solve, the solve is ignored. So the grounded should not be able to be moved anymore.

PaddleStroke, 2024-06-11:
However that does not solve the bad behavior that you have with this distance joint. Can you please provide files?

luisDias89, 2024-06-16:
@PaddleStroke
The problem with the 'joint' is described in this 'issue' - https://github.com/FreeCAD/FreeCAD/pull/14612

closed 2024-06-17 by chennes