FreeCAD/FreeCAD #14445: [Core] : Selecting a vertex with a more visible indicator, such as a ball.

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

An example of how FreeCAD is currently, without visibility:
https://github.com/FreeCAD/FreeCAD/assets/48370099/c5981e04-eb63-4cf0-af9b-13c1b3ce9aa9

An example of how it should be:
[image]

https://github.com/FreeCAD/FreeCAD/assets/48370099/c727381d-0285-41a5-9ec7-e7d37a01119c

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pure-machinery, 2024-06-05:
I would rather have just a re-scaled version of the current vertex. For example when your mouse comes near the vertex it expands and then if you click on it, it can stay scaled until the selection is cleared.

luisDias89, 2024-06-05:
@pure-machinery 
I suggested the idea because I work with SolidWorks, and it bothered me not being able to clearly see the vertex I selected. Now, whether it's with a square, a circle, or a larger color, it doesn't matter.

In the case of SolidWorks, it is a circle that, when the mouse hovers over it, shows a small ball with black borders and orange inside. When clicked (selected), it becomes a ball with black borders and blue inside.

The ball is small, but it is very visible.

benj5378, 2024-10-07:
I'm very happy with the idea of the ball