FreeCAD/FreeCAD #14446: Reduce sketcher selection from O(n^2) to O(n)

emmanuelobrien, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Selecting lots of elements in the sketcher slows down quadratically.  Repro: create an  array of ~400 vertices, area-select them with the mouse, takes about 30 seconds on my machine.

Cause: the selected objects are added one at a time, and each one invokes TaskSketcherElements.cpp : onSelectionChanged(), which de-selects and re-selects every element in the element list widget, causing the entire widget to redraw for every selected element.  (It actually iterates over all the elements twice, but it's the deselect/reselect that really slows things down).

Fix: See diff below.  Doing the deselect/reselect on just the item that was updated gets us much closer to linear time.

Context: Original approach was written by @PaddleStroke to address a bug noticed by OpenBrain:
https://github.com/FreeCAD/FreeCAD/pull/7567#issuecomment-1291870551

I could make this into a PR with a few hours work (to set up a proper fork, etc.), but it's a pretty small change so perhaps it's easier for one of the devs to do.  LMK your preference.  (Nifty element widget btw, @PaddleStroke).

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Anything else?

[speedup.txt](https://github.com/user-attachments/files/15524421/speedup.txt)

```
diff --git a/src/Mod/Sketcher/Gui/TaskSketcherElements.cpp b/src/Mod/Sketcher/Gui/TaskSketcherElements.cpp
index f4f3ed090b..c8738178a2 100644
--- a/src/Mod/Sketcher/Gui/TaskSketcherElements.cpp
+++ b/src/Mod/Sketcher/Gui/TaskSketcherElements.cpp
@@ -1387,6 +1387,8 @@ void TaskSketcherElements::onSelectionChanged(const Gui::SelectionChanges& msg)
         if (strcmp(msg.pDocName, sketchView->getSketchObject()->getDocument()->getName()) == 0
             && strcmp(msg.pObjectName, sketchView->getSketchObject()->getNameInDocument()) == 0) {
             if (msg.pSubName) {
+                ElementItem* modified_item = NULL;
+
                 QString expr = QString::fromLatin1(msg.pSubName);
                 std::string shapetype(msg.pSubName);
                 // if-else edge vertex
@@ -1399,11 +1401,13 @@ void TaskSketcherElements::onSelectionChanged(const Gui::SelectionChanges& msg)
                         int ElementId = match.captured(1).toInt(&ok) - 1;
                         if (ok) {
                             int countItems = ui->listWidgetElements->count();
+                            // Todo: Can we find the right item without iterating over the whole list?
                             for (int i = 0; i < countItems; i++) {
                                 ElementItem* item =
                                     static_cast<ElementItem*>(ui->listWidgetElements->item(i));
                                 if (item->ElementNbr == ElementId) {
                                     item->isLineSelected = select;
+                                    modified_item = item;
                                     break;
                                 }
                             }
@@ -1425,10 +1429,12 @@ void TaskSketcherElements::onSelectionChanged(const Gui::SelectionChanges& msg)
                                 ElementId, GeoId, PosId);
 
                             int countItems = ui->listWidgetElements->count();
+                            // Todo: Can we find the right item without iterating over the whole list?
                             for (int i = 0; i < countItems; i++) {
                                 ElementItem* item =
                                     static_cast<ElementItem*>(ui->listWidgetElements->item(i));
                                 if (item->ElementNbr == GeoId) {
+                                    modified_item = item;
                                     switch (PosId) {
                                         case Sketcher::PointPos::start:
                                             item->isStartingPointSelected = select;
@@ -1448,19 +1454,18 @@ void TaskSketcherElements::onSelectionChanged(const Gui::SelectionChanges& msg)
                         }
                     }
                 }
-                // update the listwidget
                 {
                     QSignalBlocker sigblk(ui->listWidgetElements);
-                    for (int i = 0; i < ui->listWidgetElements->count(); i++) {
-                        ElementItem* item =
-                            static_cast<ElementItem*>(ui->listWidgetElements->item(i));
-                        if (item->isSelected())
-                            item->setSelected(
-                                false);// if already selected, we need to reset setSelected or it
-                                       // won't draw subelements correctly if selecting several.
-                        item->setSelected(item->isLineSelected || item->isStartingPointSelected
-                                          || item->isEndPointSelected || item->isMidPointSelected);
+                    // update the listwidget
+                    if (modified_item != NULL) {
+                      if (modified_item->isSelected())
+                          modified_item->setSelected(
+                              false);// if already selected, we need to reset setSelected or it
+                                     // won't draw subelements correctly if selecting several.
+                      modified_item->setSelected(modified_item->isLineSelected || modified_item->isStartingPointSelected
+                                        || modified_item->isEndPointSelected || modified_item->isMidPointSelected);
                     }
+
                 }
             }
         }

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
Please go ahead and create a PR if you already have a solution.

PaddleStroke, 2024-06-03:
Thank you for fixing this @emmanuelobrien

emmanuelobrien, 2024-06-07:
Anything else I should do to get it reviewed and merged?  Looks like you guys are getting a lot of PRs lately.  Should I just be patient and help review other PRs?

closed 2024-06-13 by emmanuelobrien