FreeCAD/FreeCAD #14447: Broken geometry exporting a solid derived from an arc of ellipse to a step file

pyro9, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a sketch on the XZ plane.
2. Create an arc of ellipse centered on the origin. The exact dimensions don't seem to matter much, but place the major axis on the x axis.
3. I did 32 mm for the major and 12 for the minor axis. Draw the arc across the 1st quadrant from the Y axis to the X axis.
4. Close the sketch and revolve 360 degrees creating a solid.
5. Export the revolve as a step file.
6. In a new document, import the step file from step 5.

It will take a long time to complete and the solid will look like a fusion of the dome and it's mirror over the XY plane. 

When I create the arc with the major axis on the Y axis, it works normally. If I convert the arc to a B-spline, it works normally.
Creating a primitive ellipsoid and splitting it in half with a plane in the Part workbench also works.

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.55
  * ThreadProfile 1.89.0
  * Curves 0.6.8
  * 3D_Printing_Tools
  * lattice2 1.0.0
  * Silk 1.0.0
  * Assembly4 0.50.6
  * A2plus 0.4.61
  * CfdOF 1.21.6
  * fcgear.backup1677967119.4973533 1.0.0 (Disabled)
  * fcgear 1.0.0
  * sheetmetal 0.2.61
  * Render 2023.8.18
  * CurvedShapes 1.0.5
[/code]

Also tested with same version on MacOS and FreeCAD_weekly-builds-37464-conda-Linux-x86_64-py311.AppImage on the Ubuntu box.
```

### Subproject(s) affected?

None

### Anything else?

[Ellipse.zip](https://github.com/user-attachments/files/15524450/Ellipse.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
Can confirm, also when using. the PartDesign revolve.

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

pyro9, 2026-09-27:
Still relevant on 26.3 with OCC 8.0.1

```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 26.3.0dev.48937 (Git)
Build date: 2026/09/26 12:32:20
Build type: Release
Branch: PythonFreeze
Hash: 5957ff5559b9524453777740c4c0a57f8428a5f2
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 93/95.958/1
OpenGL version/vendor/renderer: 3.1 Mesa 21.2.6/Mesa/X.org/llvmpipe (LLVM 12.0.0, 256 bits)
Installed mods:
  * FourAxis
```