FreeCAD/FreeCAD #14450: Sketcher_CompModifyKnotMultiplicity tool should be removed

Roy-043, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Sketcher_CompModifyKnotMultiplicity tool seems inconsistent. It occurs next to Sketcher_BSplineIncreaseDegree and Sketcher_BSplineDecreaseDegree which are not combined in a 'Comp' tool. It would be better if the Sketcher_BSplineIncreaseKnotMultiplicity and Sketcher_BSplineDecreaseKnotMultiplicity are also separate tools in the toolbar.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-05:
I would argue that these tools ideal place are not in the toolbar at all, it would feel more natural for them to be in the right click context menu when right clicking on a bspline.

When you want to edit an object, spontaneous action is to double click it or right click it. Not to search through the toolbars.

Roy-043, 2024-06-05:
The discussion you want to start is of a different order. But these tools are already available in the 3D view context menu. The goal of this issue is address a minor, but obvious, toolbar inconsistency. IMO it is an open-and-shut case.