FreeCAD/FreeCAD #14458: [Integrated Assembly] Belt, Gear, Rack and Pinion Joints crash FreeCAD

mfro0, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

in the latest dev version, these joints crash FreeCAD (https://forum.freecad.org/viewtopic.php?p=763408#p763408)

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37577 (Git)
Build type: Release
Branch: main
Hash: 859d47bac7087c27698ff17342d71d6cf86c5432
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Installed mods: 
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * Silk 0.1.5
  * sheetmetal 0.4.14
  * Help 1.0.3
  * Assembly4 0.50.12
  * Help.backup1661594724.9387012
  * Behave-Dark-Colors.backup1661598364.718598
  * Assembly4.backup1661591923.0362408
  * DynamicData 2.62.0
  * Curves 0.6.36
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-02:
@PaddleStroke FYI

PaddleStroke, 2024-06-11:
@mfro0 thanks for the report. But could you provide reproduction steps and file please?

FEA-eng, 2024-06-12:
It seems to be working fine on Windows. Here's a redone example from the forum thread referenced above:

[assFC2V3.zip](https://github.com/user-attachments/files/15809852/assFC2V3.zip)

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git)
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-06-14:
@mfro0 is it working in the latest weekly? Please add more info, otherwise I'll close this issue.

FEA-eng, 2024-06-14:
I can confirm the crash on Ubuntu (with the same model that I shared above and that works on Windows).

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7ff90ac42520]
#1  0x7ff88f30144e in MbD::PartFrame::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1+0x1e
#2  0x7ff88f2f3217 in MbD::Part::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1+0x47
#3  /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1(+0x312bbc) [0x7ff88f312bbc]
#4  0x7ff88f35e92a in std::function<void (std::shared_ptr<MbD::Item>)> std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) from /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1+0x5a
#5  0x7ff88f35f4d6 in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1+0x76
#6  /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1(+0x120de1) [0x7ff88f120de1]
#7  0x7ff88f35fe20 in MbD::System::runPreDrag(std::shared_ptr<MbD::System>) from /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1+0x110
#8  0x7ff88f1666a2 in MbD::ASMTAssembly::runPreDrag() from /home/ubuntu/Documents/squashfs-root/usr/lib/./libOndselSolver.so.1+0x1f2
#9  0x7ff8af1eb9fb in Assembly::AssemblyObject::solve(bool, bool) from /home/ubuntu/Documents/squashfs-root/usr/lib/AssemblyApp.so+0x28b
#10  0x7ff8af1eddd3 in Assembly::AssemblyObjectPy::solve(_object*) from /home/ubuntu/Documents/squashfs-root/usr/lib/AssemblyApp.so+0x73
#11  0x7ff8af1ede54 in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) from /home/ubuntu/Documents/squashfs-root/usr/lib/AssemblyApp.so+0x24
#12  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7ff90d606068]
#13  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7ff90d5e5b5c]
#14  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7ff90d5f1532]
#15  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x181) [0x7ff90d616b71]
#16  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x21e152) [0x7ff90d61e152]
#17  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x24b) [0x7ff90d5e5b3b]
#18  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7ff90d5f1532]
#19  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x235df4) [0x7ff90d635df4]
#20  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2357b0) [0x7ff90d6357b0]
#21  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADBase.so(PP_Run_Method+0xe3) [0x7ff90c9defe3]
#22  0x7ff90c92a34b in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADBase.so+0x2b
#23  0x7ff90df8bbb3 in Gui::PythonCommand::activated(int) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x53
#24  0x7ff90df8c07e in Gui::Command::_invoke(int, bool) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0xde
#25  0x7ff90df8c604 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x134
#26  0x7ff90df8cbd5 in Gui::PythonGroupCommand::activated(int) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x425
#27  0x7ff90df8c248 in Gui::Command::_invoke(int, bool) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x2a8
#28  0x7ff90df8c604 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0x134
#29  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7ff90b2eb6f2]
#30  0x7ff90c16db7f in QActionGroup::triggered(QAction*) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5+0x2f
#31  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7ff90b2eb964]
#32  0x7ff90c16a462 in QAction::triggered(bool) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5+0x32
#33  0x7ff90c16ce2b in QAction::activate(QAction::ActionEvent) from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5+0xab
#34  /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Widgets.so.5(+0x2e12e2) [0x7ff90c2e12e2]
[... cut ...]
#77  0x7ff90b2bd212 in QCoreApplication::exec() from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libQt5Core.so.5+0x82
#78  0x7ff90de9ab27 in Gui::Application::runApplication() from /home/ubuntu/Documents/squashfs-root/usr/bin/../lib/libFreeCADGui.so+0xa27
#79  /home/ubuntu/Documents/squashfs-root/usr/bin/freecad(+0x7620) [0x56338e15b620]
#80  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7ff90ac29d90]
#81  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7ff90ac29e40]
#82  /home/ubuntu/Documents/squashfs-root/usr/bin/freecad(+0x7981) [0x56338e15b981]
```

johnsonm, 2024-06-21:
I have also encountered this. If you are looking for another reproducer, load the SingleLinkage.FCstd from the attached assembly, activate the assembly, select the top faces of the two wheels ("FakeGear" meant to try to speed up animation), and apply the Gear linkage.

[CrashGear.zip](https://github.com/user-attachments/files/15927776/CrashGear.zip)

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7fd9909d4050]
#1  0x7fd995bd17aa in __gnu_cxx::__normal_iterator<unsigned long*, std::vector<unsigned long, std::allocator<unsigned long> > >::__normal_iterator(unsigned long* const&) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x10
#2  0x7fd995bcba97 in std::vector<unsigned long, std::allocator<unsigned long> >::end() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x23
#3  0x7fd92b3c990f in MbD::PartFrame::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x27
#4  0x7fd92b3c4940 in MbD::Part::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x52
#5  /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1(+0x5ee8ad) [0x7fd92b3ee8ad]
#6  /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1(+0x5eff5f) [0x7fd92b3eff5f]
#7  /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1(+0x5efb47) [0x7fd92b3efb47]
#8  /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1(+0x5ef810) [0x7fd92b3ef810]
#9  0x7fd92b42958f in std::function<void (std::shared_ptr<MbD::Item>)>::operator()(std::shared_ptr<MbD::Item>) const from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x49
#10  0x7fd92b4282c8 in std::function<void (std::shared_ptr<MbD::Item>)> std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x4c
#11  0x7fd92b4237fd in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x8b
#12  0x7fd92b430e25 in MbD::SystemSolver::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x27
#13  0x7fd92b3eeacb in MbD::PosICNewtonRaphson::run() from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x135
#14  0x7fd92b42ff3a in MbD::SystemSolver::runPosIC() from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x80
#15  0x7fd92b4305c2 in MbD::SystemSolver::runPreDrag() from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x46
#16  0x7fd92b4231e1 in MbD::System::runPreDrag(std::shared_ptr<MbD::System>) from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0xe5
#17  0x7fd92b29166d in MbD::ASMTAssembly::runPreDrag() from /var/home/johnsonm/d/FreeCAD/build/src/3rdParty/OndselSolver/libOndselSolver.so.1+0x189
#18  0x7fd94c769a01 in Assembly::AssemblyObject::solve(bool, bool) from /var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/AssemblyApp.so+0x17d
#19  0x7fd94c7c0752 in Assembly::AssemblyObjectPy::solve(_object*) from /var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/AssemblyApp.so+0xa6
#20  0x7fd94c7bef4f in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) from /var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/AssemblyApp.so+0xf3
#21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7fd992224ba1]
#22  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7fd9921d6e8d]
#23  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7fd992164c63]
#24  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26ab4a) [0x7fd9922c5b4a]
#25  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_FastCallDictTstate+0x56) [0x7fd9921d9d76]
#26  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_Call_Prepend+0xe0) [0x7fd9921d9fa0]
#27  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1f0786) [0x7fd99224b786]
#28  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1e076e) [0x7fd99223b76e]
#29  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7fd9921d6e8d]
#30  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7fd992164c63]
#31  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26ab4a) [0x7fd9922c5b4a]
#32  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x17f6ac) [0x7fd9921da6ac]
#33  /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADBase.so(PP_Run_Method+0x16b) [0x7fd992bf4999]
[... cut ...]
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0

aiksiongkoh, 2024-06-25:
Working on it. Thanks.

aiksiongkoh, 2024-06-25:
The solver sees gear radii to be zeros. @PaddleStroke

PaddleStroke, 2024-06-26:
I will check this. However even if radii are 0 the solver should not crash. It should just ignore the joint.

aiksiongkoh, 2024-06-26:
Ok. I will check and throw exceptions.

PaddleStroke, 2024-06-27:
I can confirm the issue with CrashGear.zip document.
The issue happens if you select the 2 elements first, then trigger the tool.
If you do the other way around, trigger the tool then select elements it does not happen.

I guess if you select the element first then trigger the tool, the assembly solves before the radii are set to 1. And as the property default value is 0, and then the solver crash. So on my end I will fix the situation to avoid solving before radii are set correctly.

PaddleStroke, 2024-06-27:
So indeed this was the problem. handleInitialSelection() was doing an assembly solve before the radii were set to 1.0. I guess the initial bug report is the same, perhaps @FEA-eng if you could not reproduce on windows it's maybe because you were trying to select after starting the tool?
PR is coming shortly, if you guys can test it out.

PaddleStroke, 2024-06-27:
https://github.com/FreeCAD/FreeCAD/pull/14980
If you guys can test this PR and report if all the crashes are gone?

closed 2024-06-28 by chennes

johnsonm, 2024-06-30:
I don't know what went wrong in my prior test, but I confirm that I can now create this linkage without crash by selecting the faces and then the Gear constraint running `main` at 20f5e8f662

FredLaCasemate, 2025-01-24:
On Linux, the belt joint crashes FreeCAD if the two axes are concentric. I have not found another way to assemble two rotating concentric body with a ratio in their rotation speed.
The bug exists as well in the stable 1.0.0 version and in the 39896 built.
In the joined sample I have 3 bodies (2 pulleys ans 1 axe). If I try to add a belt joint between the 2 pulleys, Freecad crashes.

[BugBelt.FCStd.tar.gz](https://github.com/user-attachments/files/18539531/BugBelt.FCStd.tar.gz)

OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

johnsonm, 2025-01-25:
@FredLaCasemate Please open a new bug; don't just make a comment on a closed bug. Your bug report won't show for folks monitoring open bugs unless you make it a new bug report.

luzpaz, 2025-01-26:
Correct 👆