FreeCAD/FreeCAD #14464: [Assembly]: Increase produtivity on assembly workbench with import part/ insert part or body (icon menu)

luisDias89, 2024-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most of the time, when working in an assembly, if you need a new part, you will insert it externally from another file or add a new part or body, or even import from the cloud (e.g., Ondsel Lens).

The icon for importing a new component should load and have these options available, and reconfigure FreeCAD to start working immediately. For example, if you add a new body or part, it should switch the workbench to "Part Design."

Click on the "Import New Component" icon.
Options are displayed to import from a file, add a new part, or import from the cloud.
Upon selection, FreeCAD automatically switches to the appropriate workbench (e.g., "Part Design" for a new body).

Example:
[image]

What should be improved:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-05:
Sounds reasonable.
So the 'Add new part' command would do the following if I understood correctly : 
- Prompt user for a part name
- Create a new document partName.fcstd and save it
- Create a Part called partName within this new document
- Create a body within this part.
- Add a link to the part in the assembly.
- Switch to part design.

does that sounds correct?

luisDias89, 2024-06-05:
@PaddleStroke 
Yes, for "Add new part" it seems fine to me, but I would make a modification to make it more complete.

I would ask if the user wants the "Part" internally (as part of the current document) or externally (following the sequence you described).

Window with question, add with external reference?
YES:
{
Prompt the user for a part name
Create a Part called partName
Create a body within this part
Switch to part design
}
NO:
{
Prompt the user for a part name
Create a new document partName.fcstd and save it
Create a Part called partName within this new document
Create a body within this part
Add a link to the part in the assembly
Switch to part design
}

It could be a window with all this information and checkboxes, for example (just my idea).

luisDias89, 2024-06-05:
[image]

PaddleStroke, 2024-06-06:
@chennes can you please assign me please?

PaddleStroke, 2024-10-24:
I was looking into this issue, and I have one problem with it : it would create a duplicate of the 'new part' tool. Which I don't like. It feels like we would be creating an issue similar to the "there're 2 tools to create a sketch" https://github.com/FreeCAD/FreeCAD/issues/12106.

So perhaps it would make more sense to modify the existing tool?
An alternative would be to have an assembly "structure" toolbar with assembly tools. And then hide the core "structure" toolbar when in assembly workbench (as it is currently the case in sketcher edit mode)

Thoughts ? @FreeCAD/design-working-group

luisDias89, 2024-11-05:
Dear friend @PaddleStroke , I was testing the new Assembly in version 1.1.0 and came across a feature that isn’t currently included, but could be added to this menu. For example, in SolidWorks, when we add a new part, the software waits for a reference for insertion. This is incredibly useful. I've made a simple demo video to show this: I have an assembly and want to create a new part on a specific plane. I simply add the part and select the plane, and the software performs a "Join @ Mate," aligning the origin of the part with the selected plane.

https://github.com/user-attachments/assets/35977fe9-99c5-4195-9f1d-d4fc54994936

luisDias89, 2024-11-05:
@PaddleStroke How to replicate all the steps in FreeCAD? Notice the complexity involved in accomplishing the same task, and even so, the extraction of references doesn't even work:
https://github.com/user-attachments/assets/196d0ce6-aa99-4fd6-ac02-f8b77e3ba662

luisDias89, 2024-11-05:
Explanation of Inplace: https://cadbooster.com/solidworks-inplace-mate-virtual-parts/

PaddleStroke, 2024-11-05:
This is basically the top-down workflow right? If so yes it's on my todo list.

luisDias89, 2024-11-06:
@PaddleStroke  What we described on workflow is how it is inserted. This is a complement; everything is the same as in the workflow, but the part/body is inserted with a "join" called Inplace, meaning the reference is translated to a face selected by the user.
Check my videos.

luisDias89, 2024-11-06:
In summary, the Inplace Feature automates these steps shown in the video and brings the user to the point of creating a Sketch.
https://github.com/user-attachments/assets/196d0ce6-aa99-4fd6-ac02-f8b77e3ba662

Explanation of Inplace: https://cadbooster.com/solidworks-inplace-mate-virtual-parts/

pierreporte, 2024-11-06:
@PaddleStroke 

> I was looking into this issue, and I have one problem with it : it would create a duplicate of the 'new part' tool. Which I don't like. It feels like we would be creating an issue similar to the "there're 2 tools to create a sketch" https://github.com/FreeCAD/FreeCAD/issues/12106.
>
> So perhaps it would make more sense to modify the existing tool?
> An alternative would be to have an assembly "structure" toolbar with assembly tools. And then hide the core "structure" toolbar when in assembly workbench (as it is currently the case in sketcher edit mode). But imo this would generate confusion. "I don't understand Create Part is not working as it was 10 minutes ago!"

Yes I’m in favor of modifying the current new part or new assembly tool so that it’s context-aware. A new toolbar organization is going to be worked on soon so that could be part of the study.

@luisDias89 

> the software waits for a reference for insertion.

This is actually a critical feature. It is very important that the user controls where the origin of the new parts are and how it is oriented. Usually it’s either at the assembly origin (very common in aerospace) or at a selected location.

@PaddleStroke 

> This is basically the top-down workflow right? If so yes it's on my todo list.

It is heavily used in top-down workflow indeed but even bottom-up design can make use of this.

-------

Whether it creates a new file or an object directly in the current file should be dependent on the context as well. I’m working on a new file structure that will provide single-object files or current generic files. The idea is that when working with single-object files, separate files would be created, but the user would have to chose what to do when using a generic file. @PaddleStroke I’m pinging you on Discord so that you can read the preliminary discussion. I should post the results soon in github and the forum.

PaddleStroke, 2024-11-15:
Hmmm. I have started to make the tool like so : 
[image]
So a kind of customized joint creation task panel. But I think this is actually not the best idea. It's kind of messy if the user change the name or the 'New file' box.
So afterall I think it's best to just open a small dialog asking for name and 'new file' box. Then open a create joint task panel prefilled with the new part.

luisDias89, 2024-11-15:
@PaddleStroke  Looks good
What can be selected in this ComboBox?

closed 2024-12-02 by yorikvanhavre